cocotb/cocotb #2538: Fix private CI

imphil, 2021-05-03:
Currently private CI fails on master as well as on branches. Fix that.

imphil, 2021-05-03:
Initially the disk of the runner was full after a failed attempt to create more logging. That is resolved now. But now tests run into our 20 minute timeout. They finished in around 6 minutes before. I'll continue looking.

cmarqu, 2021-05-03:
Thanks for taking a look, I kept meaning to debug this but didn't find enough time.

imphil, 2021-05-03:
Private CI is now working again.

* A bit of filesystem caching logic wasn't properly set up, causing more network traffic than expected, slowing down builds.
* The latest Azure Pipelines agent consumes a full CPU for no good reason, reverting to an older version does the trick for now.

That leaves us with one problem: aa849aa doubled build time roughly. I created https://github.com/cocotb/cocotb/pull/2540 to back it out for now and unblock master, and we can then work on a less resource-intensive solution without time pressure (e.g. #2531).

(Side note: we run on AWS t3a burstable instances; meaning if we have many CI runs after each other they will have less CPU available than if we just run a single job once every couple of hours, as we do in "normal operation". Keep that in mind when comparing build times for old CI runs.)

closed 2021-05-03 by imphil