cocotb/cocotb #2543: assignment at end of test isn't complete by start of next test

mattvenn, 2021-05-11:
cocotb 1.4.0
ubuntu 20.04
python 3.8.5
iverilog 12.0

expectation: an assignment at the end of a test should be complete by the start of the next test.

```
@cocotb.test()
async def test1(dut):
    clock = Clock(dut.clk, 10, units="us")
    cocotb.fork(clock.start())
    dut.test <= 1

@cocotb.test()
async def test2(dut):
    clock = Clock(dut.clk, 10, units="us")
    cocotb.fork(clock.start())
    assert dut.test == 1
```

The assertion in test2 fails.

[assign-mcve.zip](https://github.com/cocotb/cocotb/files/6458407/assign-mcve.zip)

mattvenn, 2021-05-11:
I have been assuming that I can split a test up into multiple sections and they run in order.

Then that's useful when I build other tests later by importing the useful bits. So for example I always have something that does a reset but I usually want to call that one time at the start of a set of tests.

eric-wieser, 2021-05-11:
Right, it's true that they run in order and you can rely on that - but I think that was done primarily for determinism and not with the specific intent of enabling multi-part tests. Certainly the `TESTCASE` command line argument is not going to play well with your approach.

That's not to say your approach is a bad one - I just don't think it was designed in as a deliberate feature.

closed 2021-06-11 by mattvenn