cocotb/cocotb #2548: Customization of results.xml with "factory.add_optiom(..."

henriquedemoura, 2021-05-18:
Hello! I'd like to suggest a feature to the results.xml file that we get at the end of a simulation.

I'm running some tests permutations with a code that looks like this:
```
for test in [test1, test2]:
        factory = TestFactory(test)
        factory.add_option("config1",            [True, False])
        factory.add_option("config2",            ["opt-a", "opt-b", "opt-c"])
        factory.add_option("config3",            [True, False])
        factory.generate_tests()
```

Then in my results.xml, I got the list of all the test permutations with this setting.
Right now it is only possible to get the name of the test, if it has passed/failed, its simulation time, its real time, and its ratio.

Could you guys add a way to add to each test in the results.xml the value of the custom options?

eric-wieser, 2021-05-18:
Doesn't the name of the test contain the option information?

henriquedemoura, 2021-05-18:
When the tests are generated, they get a number after their names, and that's that.
In the example I gave, my results.xml looks like this:

```xml
<testsuites name="results">
  <testsuite name="all" package="all">
    <property name="random_seed" value="1621359523" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_001" ratio_time="3610.792267049259" sim_time_ns="18.001" time="0.004985332489013672" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_002" ratio_time="3007.7948491753646" sim_time_ns="18.001" time="0.005984783172607422" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_003" ratio_time="2256.070827227634" sim_time_ns="18.000999999999998" time="0.00797891616821289" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_004" ratio_time="4512.141654455269" sim_time_ns="18.001000000000005" time="0.003989458084106445" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_005" ratio_time="3591.5548617638647" sim_time_ns="18.00099999999999" time="0.005012035369873047" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_006" ratio_time="3610.446934965571" sim_time_ns="18.001000000000005" time="0.004985809326171875" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_007" ratio_time="4511.602408365702" sim_time_ns="18.001000000000005" time="0.0039899349212646484" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_008" ratio_time="4482.939455171596" sim_time_ns="18.001000000000005" time="0.004015445709228516" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_009" ratio_time="3008.3940831175" sim_time_ns="18.000999999999976" time="0.005983591079711914" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_010" ratio_time="3610.2742936929194" sim_time_ns="18.001000000000005" time="0.0049860477447509766" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_011" ratio_time="4512.141654455269" sim_time_ns="18.001000000000005" time="0.003989458084106445" />
    <testcase classname="module_test" file="C:\Users\henri\Anaconda3\lib\site-packages\cocotb\regression.py" lineno="604" name="test_012" ratio_time="3608.721264888635" sim_time_ns="18.001000000000005" time="0.004988193511962891" />
  </testsuite>
</testsuites>
``` 

After looking in line 481 of cocotb/regression.py I found the function `_log_test_summary()` that generates this file. It looks to me that the options are ignored because the generation of the XML is static.

ktbarrett, 2021-05-18:
What do pytest tests marked with `pytest.mark.parameterize` output in XML? Is there a some standard output for test parameterization in xUnit?

eric-wieser, 2021-05-18:
I'm not sure, but pytest generates much better test names. Perhaps just doing that is enough to solve the problem, by changing this line:

https://github.com/cocotb/cocotb/blob/b862e096249f1043a1be1958bb56ce8de0fc5634/cocotb/regression.py#L711

ktbarrett, 2021-06-19:
Changing this would change the test name and break people using `TestFactory` and `TESTCASE` together. Still I see no reason to not change it.

ktbarrett, 2021-06-19:
Using pytest style test names would also not work easily because test arguments are passed by kwargs and not by position. We could add code to figure out order, or we could just log the name using the repr of the kwargs dict: `test_name{"arg1": 4, "arg2": 0}`. I'm also wondering how this would work with `TESTCASE`. I think you would need to change `TESTCASE` to support wildcards or a regex.

ktbarrett, 2021-06-19:
Also, if we change how tests are named, the `prefix` and `postfix` arguments to `generate_tests` can be deprecated as they will serve no purpose where test names will no longer be numbered.

closed 2021-06-22 by ktbarrett

f-cozzocrea, 2022-12-13:
Is this feature being planned for a future release? Looks like it was planned for 1.6, then reverted. 
This feature would be extremely useful for testbenches that I have with hundreds of iterations.

PR #2566 (2021-06-22): Change generated names of TestFactory tests
Closes #2548.

Previously the test name did not reflect the options passed and were
simply incrementing numbers appended to the test name. This made it hard
to determine from the results.xml what options were passed to any test:
`test_name_001`, `test_name_002`, etc.

This change formats the name like a function call with named arguments:
`test_name(arg1="arg", arg2=6)`, `test_name(arg1="", arg2=101)`, etc.

See [here](https://github.com/cocotb/cocotb/runs/2866967563?check_suite_focus=true#step:25:762).