cocotb/cocotb #2553: checkpoints file not working properly,does anyone know what is the problem??

karimtamer23, 2021-05-29:
`from cocotb import simulator
 from cocotb.handle import *

checkpoint_hier = []

#this function must be called prior to creating any checkpoint
def get_checkpoint_hier(entity):
iterator = simulator.iterate(entity._handle,simulator.OBJECTS)
while True:
try:
ii = simulator.next(iterator)
except StopIteration:
break

    hdl = SimHandle(ii, entity._path + "." + simulator.get_name_string(ii))
    
    if ((simulator.get_type(ii) is simulator.MODULE) or
      (simulator.get_type(ii) is simulator.NETARRAY) or
      (simulator.get_type(ii) is simulator.GENARRAY)):
        get_checkpoint_hier(hdl)
    elif simulator.get_type(ii) is simulator.REG:
        checkpoint_hier.append(hdl)

#returns a created checkpoint
def checkpoint():
checkpoint_data = {}
for reg in checkpoint_hier:
checkpoint_data[reg] = reg.value

return checkpoint_data

#restores a checkpoint
def restore(checkpoint_data):
for reg in checkpoint_data:
reg.setimmediatevalue(checkpoint_data[reg])`
_____________________________________________________________________________________________
This is what i get when i run the simulation :

Test Failed: test_tree (result was AttributeError)
Traceback (most recent call last):
File "/home/karim/apbi2c_cocotb_example/tb/test_i2c.py", line 281, in test_tree
get_checkpoint_hier(dut)
File "/home/karim/apbi2c_cocotb_example/tb/checkpoint.py", line 48, in get_checkpoint_hier
iterator = simulator.iterate(entity._handle,simulator.OBJECTS)
AttributeError: module 'cocotb.simulator' has no attribute 'iterate'

karimtamer23, 2021-05-29:
@eric-wieser do you know what might be the problem?

eric-wieser, 2021-05-29:
Yes, the functions `simulator.iterate` and `simukator.next` no longer exist, I think we replaced them with a simpler API.

eric-wieser, 2021-05-29:
It's now `for ii in hdl.iterate(OBJECTS)`:

https://docs.cocotb.org/en/stable/library_reference.html#cocotb.simulator.gpi_sim_hdl.iterate

karimtamer23, 2021-05-29:
> It's now `for ii in hdl.iterate(simulator.OBJECTS)`:
> 
> https://docs.cocotb.org/en/stable/library_reference.html#cocotb.simulator.gpi_sim_hdl.iterate

can you tell me which part would that replace in the old code as i am still new to cocotb and cant figure out how to use it?
Thanks in advance

eric-wieser, 2021-05-29:
You should replace the part that the error message tells you is broken, `get_checkpoint_hier`!

eric-wieser, 2021-05-29:
Also, the indents are all wrong in your code; I assume something went wrong when you pasted into github?

karimtamer23, 2021-05-29:
> Also, the indents are all wrong in your code; I assume something went wrong when you pasted into github?

yes, when i pasted the code into github it got messed up

> You should replace the part that the error message tells you is broken, `get_checkpoint_hier`!
okay so would this fix it?

`def get_checkpoint_hier(entity):
  
     for ii in hdl.iterate(simulator.OBJECTS):
    	
    #iterator  = simulator.iterate(entity._handle,simulator.OBJECTS)
    #while True:
     #   try:
      #      ii = simulator.next(iterator)
       # except StopIteration:
        #    break

        hdl = SimHandle(ii, entity._path + "." + simulator.get_name_string(ii))
        
        if ((simulator.get_type(ii) is simulator.MODULE) or
          (simulator.get_type(ii) is simulator.NETARRAY) or
          (simulator.get_type(ii) is simulator.GENARRAY)):
            get_checkpoint_hier(hdl)
        elif simulator.get_type(ii) is simulator.REG:
            checkpoint_hier.append(hdl)`

eric-wieser, 2021-05-29:
Why don't you try it and see? Github will stop messing with your code if you type it as

````
```python
Your code here
```
````

karimtamer23, 2021-05-29:
```python
checkpoint_hier = []

#this function must be called prior to creating any checkpoint
def get_checkpoint_hier(entity):
   for ii in entity._handle.iterate(simulator.OBJECTS):
    	
      hdl = SimHandle(ii, entity._path + "." + entity._handle.get_name_string())
        
      if ((entity._handle.get_type() is simulator.MODULE) or
          (entity._handle.get_type() is simulator.NETARRAY) or
          (entity._handle.get_type() is simulator.GENARRAY)):
            get_checkpoint_hier(hdl)
      elif entity._handle.get_type() is simulator.REG:
            checkpoint_hier.append(hdl)
            
```
Okay,so this is what i got so far.the test now runs successfully with no errors but gives this warning : "VPI: Implementation does not know how to iterate over vpiReg(48)" more than one time.Not sure if this causes any problems or not.i tried searching it up but couldnt find anything.

eric-wieser, 2021-05-29:
You might find that using `for ii in entity` works better - it will construct the SimHandles for you too. I don't know whether there is a way to avoid that warning, it may be a problem with your simulator

karimtamer23, 2021-05-30:
> You might find that using `for ii in entity` works better - it will construct the SimHandles for you too. I don't know whether there is a way to avoid that warning, it may be a problem with your simulator

What will ii represent in that case?keeping in mind that "entity" represents my "dut"

ktbarrett, 2021-06-01:
This isn't an "issue" with the cocotb code base, so I'm converting to a discussion.

closed 2021-06-01 by ktbarrett