cocotb/cocotb #2555: Beeing able to see an array of interfaces from dut in SV

carrecuir1, 2021-06-02:
Hey all ! I'm using cocotb 1.5.2 with Verilator 4.106 and I'm trying to see the from my dut my array of interfaces. But, I've got an error seeing that it can't find my array. Here's my code : 

**My top_layer**
`module dummy_tb;

    dummyInterface #(32)  dum_if[1:0]();
    logic                           clk;
    logic                           reset;
    logic [63 : 0]                  o_data[2];
    dummy #(2, 64) dummy_dut(.dummy(dum_if), .clk, .reset, .o_data);

endmodule`

**My module to be tested**
`module dummy #(parameter
                NUMBER_CHANNEL = 2,
                DATA_LENGTH = 64)
    (// Ports declaration
        dummyInterface.internal  dummy[NUMBER_CHANNEL - 1:0],
        input logic                                   reset,
        input logic                                   clk,
        output logic [DATA_LENGTH - 1:0]              o_data [NUMBER_CHANNEL]       
    );

    logic [DATA_LENGTH - 1:0]                   o_data_r[NUMBER_CHANNEL];
    
    generate
        for (genvar i = 0; i < NUMBER_CHANNEL; i++) begin : dummy_block
            assign o_data_r[i] = {dummy[i].data_1, dummy[i].data_2};
        end
    endgenerate

    assign o_data = o_data_r;
    
endmodule // dummy`

**My python folder**
`import cocotb
from cocotb.triggers import Timer
from cocotb.clock import Clock
import numpy as np
import random

@cocotb.test()
async def dummy_test(dut):
    
    cocotb.fork(Clock(dut.clk, 10, units='ns').start())

    interfaces = dut.dum_if`
    
The simulation says : "AttributeError: dummy_tb contains no object named dum_if"

I'm also using gtkwave and here's the modules seen : 

[image]

So here's my question : Is it possible to see the array of interface using cocotb and Verilator ?

ktbarrett, 2021-06-02:
Support for interfaces is located in the SV VPI extensions. Verilator does not support those extensions yet and neither do we. They will have to support this first before we can build against it.

carrecuir1, 2021-06-02:
Ah I see, thanks a lot for your response.

closed 2021-06-02 by carrecuir1