cocotb/cocotb #2560: Make Event an Awaitable[None]

ktbarrett, 2021-06-10:
Make `cocotb.triggers.Event` an `Awaitable[None]`. Supply an `__await__` implementation that proxies `event.wait()` and returns `None`.

ktbarrett, 2021-06-10:
Looks like `asyncio.Event` is not an `Awaitable`. [See upstream issue.](https://bugs.python.org/issue33544) Once again, it comes down to `asyncio` trying to mimic the threading libraries and not for any good reason. I don't have their same reservations about supporting both patterns, but I would like interface compatibility with `asyncio` to some extent.

eric-wieser, 2021-06-10:
I think the fact that a core python maintainer thinks this spelling is a bad idea (from your link), and the fact that even trio [does not allow this](https://github.com/python-trio/trio/blob/6b313de9edf3212c6e5ccfa69c68ef8a47bafc2f/trio/_sync.py#L60) means we should not do it either.

I don't think we should diverge from convention just because we feel like it.

I don't think cco

ktbarrett, 2021-06-10:
Not supporting this decreases the ability to write generic code; requiring the development of a wrapper to get the same effect (see below). On the contrary, requiring users to type `wait` brings no value to *anyone*. There is no real argument in the thread against adding `__await__`, mostly just against the original suggestion to deprecate `wait`. 

```python
class AwaitableEvent(Awaitable, Event):

    def __await__(self):  # this is asinine
        yield self.wait()
```

eric-wieser, 2021-06-10:
What type of generic code are you unable to write without this feature?

"If we don't include it then the user is forced to add it themselves" isn't enough to justify a feature, because it can be used to justify anything you like!

ktbarrett, 2021-06-10:
Anything that could accept an Awaitable could take a cocotb coroutine, an `async def` coroutine, a `RunningTask`, a `Future`, and so many other thing, besides Events.

> isn't enough to justify a feature, because it can be used to justify anything you like!

> Having *both* 'await event.wait()' and 'await event' worries me.  IMO the slight readability improvement isn't worth the added complexity.
> worries me
> "complexity"

Isn't either...

eric-wieser, 2021-06-10:
> Anything that could accept an Awaitable

You can always pass `e.wait()` to such an API today.

If you don't agree with the decision made by Yury in the bpo issue, can I suggest you add a reply there? I imagine "complexity" refers to users having to read two different ways of doing things rather than complexity of implementation.

ktbarrett, 2021-06-20:
> You can always pass e.wait() to such an API today.

You would have to know it's an Event, meaning a special case would have to be added to handle them. Doing so also removes the rest of the Event interface from the object, so downcasting Awaitable values is not possible. It's simply an anachronism, it hurts the ability to write generic code with Awaitable-like types.

No response from core devs. 

I'll close the PR since I don't see this going anywhere soon and it's trivial to implement. But I'll leave the issue open to see if we get any more comments one way or the other.

eric-wieser, 2021-06-20:
> No response from core devs.

I don't see a response from you on the issue you linked. Did you ask via another channel?

ktbarrett, 2021-06-20:
I opened [another bug report](https://bugs.python.org/issue44373) since the original bug report included the suggestion of deprecating `wait()`, which I don't think they would ever agree to.

eric-wieser, 2021-06-20:
Thanks for filing that bpo issue and closing the PR. I'm happy to keep this issue open for now, especially while your new issue on bpo is open.

> You would have to know it's an `Event`, meaning a special case would have to be added to handle them.

I don't understand this argument - if you find yourself trying to add this special case, it means that _your_ caller forgot to call `.wait()` before passing you the object.

closed 2025-02-21 by ktbarrett