cocotb/cocotb #2563: Add access to preponed region of clock edge for monitors in Triggers

gtaylormb, 2021-06-14:
The monitors I've seen written in cocotb (including ones in cocotb-bus) have been written to sample in the postpone region of the clock edge accessed by calling ReadOnly() after RisingEdge(), when all signals are assumed to have settled and no more updates may occur on the clock edge event. It seems this is the intended pattern for monitors written in cocotb.

There are two things wrong about this:
1. When an error is detected in the monitor, it appears to occur a clock edge early. This is incorrect and confusing when debugging, but is mostly tolerable once you are aware of this behavior.
2. More importantly, it does not allow for simulation models that introduce any delays or back-annotated timing simulations. These would update after the clock edge, but typically long after the postpone region of ReadOnly() occurred. The assumption that no more updates will occur on signals after the clock edge with ReadOnly() is incorrect for anything other than purely behavioral simulation models. Of course in the real circuit, with propagation delays through combinational logic any changes are allowed to occur up until the setup time of the next clock edge.

SystemVerilog has defined the prepone region, what they refer to as #1step, which occurs just before the clock edge and *before* anything signals are allowed to update. Concurrent assertions fire in this region by default, and you may access this region in simulation monitors by using a clocking block and #1step on the inputs. Sampling in the prepone region (or something equivalent) fixes both of the issues above.

Both of these issues here would be solved with this as well: https://github.com/cocotb/cocotb/issues/1060#issuecomment-519527668 and https://github.com/cocotb/cocotb/issues/204

eric-wieser, 2021-06-14:
#1662 is likely relevant here

ktbarrett, 2021-06-14:
@gtaylormb Is this a general advisory? It seems like this issue belongs in cocotb-bus, where the problematic code exists, not here. I can transfer.

gtaylormb, 2021-06-14:
@ktbarrett is there a way to access the preponed region (or equivalent) already in cocotb? If so, cocotb-bus and all other monitor examples should be updated, but if it does not exist I think the functionality needs to be added to cocotb first.

ktbarrett, 2021-06-14:
@gtaylormb `RisingEdge(clk)` leaves you after clock changes, but no sympathetic signals have changed. Sampling any signal here will give you the previous values. This is equivalent to a registered process.

@eric-wieser I think the issue is that the cocotb-bus Monitors are sampling the transaction in the same clock cycle as the next transaction is applied by (ab)using `ReadOnly` and `ReadWrite`, rather than when the transaction is actually seen by the HDL (which is on the next clock edge typically). I don't write my monitors this way, and it is not good practice IMO.