cocotb/cocotb #2564: Improving cocotb output formatting

ktbarrett, 2021-06-17:
Following on #2436.

Right now, test outputs looks like this.

[image]

Notice the following:
* Most things aren't highlighted
* Starting the test `Running test: ...` is BG highlighted blue
* `Test Passed` is not highlighted
* `Test Failed` is FG highlighted red
* FAIL line in summary is BG highlighted red (and the coloring includes the newline =/)
* The final summary is also not aligned with the per-test summary

Maybe it's just me, but I think this could be improved. What are your opinions?

ktbarrett, 2021-06-17:
Another thing that could be improve may be startup and shutdown messages.

cmarqu, 2021-06-17:
Slightly related: #2320

cmarqu, 2021-06-17:
Maybe we could collect example output of other programs. AFL is one such example that I like: https://www.youtube.com/watch?v=uj1FTiczJSE&t=820s

ktbarrett, 2021-06-20:
It seems the failure red BG highlighting extending passed the end of the summary is an issue specific to Windows Terminal and likely the line-endings support.

I propose we merge the summary into the PASS/FAIL table. A final line of the PASS/FAIL table could report totals: Somehting like this:

```
     0.00ns INFO     ********************************************************************************
                     ** TEST                    PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                     ********************************************************************************
                     ** test_module.test_pass     PASS            0.00          0.00         1.99  **
                     ** test_module.test_fail     FAIL            0.00          0.00         1.99  **
                     **                            1/1            0.00          0.00         0.99  **
                     ********************************************************************************
```

ktbarrett, 2021-06-20:
Perhaps we can also list skipped tests as `SKIP` rather than `N/A` and count those towards the total?

ktbarrett, 2021-07-19:
I think between #2618 and #2619, this has been accomplished.

closed 2021-07-19 by ktbarrett