cocotb/cocotb #2568: Inconsistent BinaryValue bigEndian representation

o-alexandre-felipe, 2021-06-20:
Maybe relevant for #2059

Starting with
```
from cocotb.binary import *
UNSIGNED=BinaryRepresentation.UNSIGNED
SIGNED_MAGNITUDE=BinaryRepresentation.SIGNED_MAGNITUDE
TWOS_COMPLEMENT=BinaryRepresentation.TWOS_COMPLEMENT
```

```python
[int(BinaryValue(i, n_bits=4, bigEndian=True, binaryRepresentation=UNSIGNED)) for i in range(16)]```

 Would give
```[0, 8, 8, 12, 8, 10, 12, 14, 8, 9, 10, 11, 12, 13, 14, 15]```

Similarly for `SIGNED_MAGNITUDE` and `TWOS_COMPLEMENT` would produce `int(BinaryValue(x)) != x`.

benreynwar, 2023-11-18:
What's the status with this?  Just spent a few hours tracking down a bug and ended up discovering some BinaryValue weirdness.
```
>>> from cocotb.binary import BinaryValue
>>> bv = BinaryValue(n_bits=5)
>>> bv.integer = 3
>>> bv.integer
24
```
The issue discussion above from a couple of years ago suggested that BinaryValue was being replaced.  Is that still the plan?

ktbarrett, 2023-11-18:
Yes. There is an open PR. It should be completed before 2.0 is released.

Ans thanks for bringing this issue to my attention.

closed 2023-11-18 by ktbarrett