cocotb/cocotb #2571: Cocotb installation issue

Future-progrmer, 2021-06-21:
Hello ,
I am facing issues installing cocotb as I am checking for cocotb -config --version , I have following enviroment specs.
* windows 10 (64 bit),
* I have installed icarus verilog (64 bit),
* I am using Pycharm with python version 3.9.5 .
The Error showing is :
$ make
Traceback (most recent call last):
  File "c:\python\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\python\Scripts\cocotb-config.exe\__main__.py", line 4, in <module>
  File "c:\python\lib\site-packages\cocotb\__init__.py", line 43, in <module>
    import cocotb.handle
  File "c:\python\lib\site-packages\cocotb\handle.py", line 38, in <module>
    from cocotb import simulator
ImportError: DLL load failed while importing simulator: The specified module could not be found.
Makefile:47: /Makefile.sim: No such file or directory
make: *** No rule to make target '/Makefile.sim

ktbarrett, 2021-06-21:
Are there DLLs in `c:\python\lib\site-packages\cocotb\libs\`?

Future-progrmer, 2021-06-22:
yes 

> 
> 
> Are there DLLs in `c:\python\lib\site-packages\cocotb\libs\`?
Yes I have checked , there are files name cocotb.dll and many more

themperek, 2021-06-22:
So the problem is already running `cocotb-config` ?

Is doing this also an issue: `python -c "import cocotb"` ?

Do you have `c:\python\lib\site-packages\cocotb\simulator.*.dll` ?

ktbarrett, 2021-06-22:
@Future-progrmer could you open up `c:\python\lib\site-packages\cocotb\__init__.py` and add the following lines at the very top and try again?

```python
import pathlib, os
libdir = path.Path(__file__).parent / "libs"
os.add_dll_directory(libdir)
```

```
ImportError: DLL load failed while importing simulator: The specified module could not be found.
```

This isn't a `ModuleNotFoundError`, `simulator.dll` is found, but it is failing to import because dependent modules (in the lib directory) are not being found.

Future-progrmer, 2021-06-22:
> 
> 
> So the problem is already running `cocotb-config` ?
> Actually am not able to run tests on the examples of the repository either am doing wrong or I dont know.
> Is doing this also an issue: `python -c "import cocotb"` ?
> This is not working...
> Do you have `c:\python\lib\site-packages\cocotb\simulator.*.dll` ?
> No, there isn't ant file named simulator.*.dll in the following path

Future-progrmer, 2021-06-22:
> 
> 
> @Future-progrmer could you open up `c:\python\lib\site-packages\cocotb\__init__.py` and add the following lines at the very top and try again?
> 
> ```python
> import pathlib, os
> libdir = path.Path(__file__).parent / "libs"
> os.add_dll_directory(libdir)
> ```
> 
> ```
> ImportError: DLL load failed while importing simulator: The specified module could not be found.
> ```
> Okay yes this is fixed now. 
> This isn't a `ModuleNotFoundError`, `simulator.dll` is found, but it is failing to import because dependent modules (in the lib directory) are not being found.

> But again am trying to run the c:\cocotb-master\examples\simple_dff this test using make this is  showing ,

Asus@DESKTOP-9R8GN8D MINGW64 /c/cocotb-master/examples/simple_dff (master)
$ make Sim=icaurs
Traceback (most recent call last):
  File "c:\python\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\python\Scripts\cocotb-config.exe\__main__.py", line 4, in <module>
  File "c:\python\lib\site-packages\cocotb\__init__.py", line 34, in <module>
    libdir = path.Path(__file__).parent / "libs"
NameError: name 'path' is not defined
Makefile:15: /Makefile.sim: No such file or directory
make: *** No rule to make target '/Makefile.sim'.  Stop.

ktbarrett, 2021-06-22:
@Future-progrmer 

Sorry, typo...
```python
import pathlib, os
libdir = pathlib.Path(__file__).parent / "libs"
os.add_dll_directory(libdir)
```

Future-progrmer, 2021-06-22:
> 
> 
> > @Future-progrmer could you open up `c:\python\lib\site-packages\cocotb\__init__.py` and add the following lines at the very top and try again?
> > ```python
> > import pathlib, os
> > libdir = path.Path(__file__).parent / "libs"
> > os.add_dll_directory(libdir)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ```
> > ImportError: DLL load failed while importing simulator: The specified module could not be found.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Okay yes this is fixed now.
> > This isn't a `ModuleNotFoundError`, `simulator.dll` is found, but it is failing to import because dependent modules (in the lib directory) are not being found.
> 
> > But again am trying to run the c:\cocotb-master\examples\simple_dff this test using make this is  showing ,
> 
> > Asus@DESKTOP-9R8GN8D MINGW64 /c/cocotb-master/examples/simple_dff (master)
> > $ make Sim=icaurs
> > Traceback (most recent call last):
> > File "c:\python\lib\runpy.py", line 197, in _run_module_as_main
> > return _run_code(code, main_globals, None,
> > File "c:\python\lib\runpy.py", line 87, in _run_code
> > exec(code, run_globals)
> > File "C:\python\Scripts\cocotb-config.exe__main__.py", line 4, in 
> > File "c:\python\lib\site-packages\cocotb__init__.py", line 34, in 
> > libdir = path.Path(**file**).parent / "libs"
> > NameError: name 'path' is not defined
> > Makefile:15: /Makefile.sim: No such file or directory
> > make: *** No rule to make target '/Makefile.sim'.  Stop.

> 
> 
> > @Future-progrmer could you open up `c:\python\lib\site-packages\cocotb\__init__.py` and add the following lines at the very top and try again?
> > ```python
> > import pathlib, os
> > libdir = path.Path(__file__).parent / "libs"
> > os.add_dll_directory(libdir)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ```
> > ImportError: DLL load failed while importing simulator: The specified module could not be found.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Okay yes this is fixed now.
> > This isn't a `ModuleNotFoundError`, `simulator.dll` is found, but it is failing to import because dependent modules (in the lib directory) are not being found.
> 
> > But again am trying to run the c:\cocotb-master\examples\simple_dff this test using "make"this is  showing ,
> 
> > Asus@DESKTOP-9R8GN8D MINGW64 /c/cocotb-master/examples/simple_dff (master)
> > $ make Sim=icaurs
> > Traceback (most recent call last):
> > File "c:\python\lib\runpy.py", line 197, in _run_module_as_main
> > return _run_code(code, main_globals, None,
> > File "c:\python\lib\runpy.py", line 87, in _run_code
> > exec(code, run_globals)
> > File "C:\python\Scripts\cocotb-config.exe__main__.py", line 4, in 
> > File "c:\python\lib\site-packages\cocotb__init__.py", line 34, in 
> > libdir = path.Path(**file**).parent / "libs"
> > NameError: name 'path' is not defined
> > Makefile:15: /Makefile.sim: No such file or directory
> > make: *** No rule to make target '/Makefile.sim'.  Stop.

ktbarrett, 2021-06-22:
> No, there isn't ant file named simulator.*.dll in the following path

That's wrong... Have you tried uninstalling and reinstalling cocotb?

Future-progrmer, 2021-06-22:
> 
> 
> > No, there isn't ant file named simulator.*.dll in the following path
> 
> That's wrong... Have you tried uninstalling and reinstalling cocotb?

>Actually am finding very tough to install it from documentation and youtube doesnt have any proper tutorial ,i can reinstall it but I think  I am going wrong way..it would be great if u willing to help.

Thank you

Future-progrmer, 2021-06-22:
> 
> 
> @Future-progrmer
> 
> cocotb is uninstalled by calling `pip uninstall cocotb` and is installed by calling `pip install cocotb`. If there are errors in the installation process, feel free to post them here.

> I have installed Gitbash in order to install "make". I should write there only right?

> Also as i have mentioned earlier I have Pycharm as my IDE and installed cocotb there also as package, will it cause any problem or I should just reinstall in gitbash.

Future-progrmer, 2021-06-22:
> 
> 
> @Future-progrmer
> 
> cocotb is uninstalled by calling `pip uninstall cocotb` and is installed by calling `pip install cocotb`. If there are errors in the installation process, feel free to post them here.

> Its showing this now, 

>Asus@DESKTOP-9R8GN8D MINGW64 ~/Desktop
$ pip uninstall cocotb
Found existing installation: cocotb 1.5.2
Uninstalling cocotb-1.5.2:
  Would remove:
    c:\python\lib\site-packages\cocotb-1.5.2.dist-info\*
    c:\python\lib\site-packages\cocotb\*
    c:\python\scripts\cocotb-config.exe
Proceed (y/n)? y
  Successfully uninstalled cocotb-1.5.2

Asus@DESKTOP-9R8GN8D MINGW64 ~/Desktop
$ pip install cocotb
Collecting cocotb
  Using cached cocotb-1.5.2-cp39-cp39-win_amd64.whl
Requirement already satisfied: cocotb-bus<1.0 in c:\python\lib\site-packages (from cocotb) (0.1.1)
Installing collected packages: cocotb
Successfully installed cocotb-1.5.2
WARNING: You are using pip version 21.1.1; however, version 21.1.2 is available.
You should consider upgrading via the 'c:\python\python.exe -m pip install --upgrade pip' command.

Asus@DESKTOP-9R8GN8D MINGW64 ~/Desktop
$ cocotb -config --version
bash: cocotb: command not found

ktbarrett, 2021-06-22:
@Future-progrmer `cocotb-config` is a single word. I also doubt you will be able to call `cocotb-config` directly since Windows does not add the `Scripts` directly to the path. If that command does not work, try `python -m cocotb.config --version`.

Future-progrmer, 2021-06-22:
> 
> 
> @Future-progrmer `cocotb-config` is a single word. I also doubt you will be able to call `cocotb-config` directly since Windows does not add the `Scripts` directly to the PATH. If that command does not work, try `python -m cocotb.config --version`.

> Now its showing This after reinstalling and copy pasting the about fix for :-
>"ImportError: DLL load failed while importing simulator: The specified module could not be found."
>Asus@DESKTOP-9R8GN8D MINGW64 ~/Desktop
$ python -m cocotb.config --version
Traceback (most recent call last):
  File "C:\python\lib\runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "C:\python\lib\runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "C:\python\lib\site-packages\cocotb\__init__.py", line 34, in <module>
    libdir = path.Path(__file__).parent / "libs"
NameError: name 'path' is not defined

> now its showing this name error

ktbarrett, 2021-06-22:
@Future-progrmer See https://github.com/cocotb/cocotb/issues/2571#issuecomment-866023777. I typo'd the code I sent to you to add to `__init__.py`. Replace it with what is in the linked comment.

Future-progrmer, 2021-06-23:
> 
> 
> @Future-progrmer See [#2571 (comment)](https://github.com/cocotb/cocotb/issues/2571#issuecomment-866023777). I typo'd the code I sent to you to add to `__init__.py`. Replace it with what is in the linked comment.

> This comes after I replace the correct code

>Asus@DESKTOP-9R8GN8D MINGW64 ~/Desktop
$ python -m cocotb.config --version
Traceback (most recent call last):
  File "C:\python\lib\runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "C:\python\lib\runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "C:\python\lib\site-packages\cocotb\__init__.py", line 46, in <module>
    import cocotb.handle
  File "C:\python\lib\site-packages\cocotb\handle.py", line 38, in <module>
    from cocotb import simulator
ImportError: DLL load failed while importing simulator: The specified module could not be found.

ktbarrett, 2021-06-23:
@Future-progrmer Did you confirm there is now a `c:\python\lib\site-packages\cocotb\simulator.*.dll` file after re-installing? That file has to exist for any other change to matter. If it doesn't exist it could be that pip is reusing a corrupted wheel. Try uninstalling again and reinstalling, but passing the `--no-cache-dir` option.

```
> pip uninstall cocotb
> pip install --no-cache-dir cocotb
```

Future-progrmer, 2021-06-23:
> 
> 
> @Future-progrmer Did you confirm there is now a `c:\python\lib\site-packages\cocotb\simulator.*.dll` file after re-installing? That file has to exist for any other change to matter. If it doesn't exist it could be that pip is reusing a corrupted wheel. Try uninstalling again and reinstalling, but passing the `--no-cache-dir` option.
> 
> ```
> > pip uninstall cocotb
> > pip install --no-cache-dir cocotb
> ```
> This comes now 

>Asus@DESKTOP-9R8GN8D MINGW64 ~/Desktop
$ pip install --no-cache-dir cocotb
Collecting cocotb
  Downloading cocotb-1.5.2.tar.gz (579 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'done'
Requirement already satisfied: cocotb-bus<1.0 in c:\python\lib\site-packages (from cocotb) (0.1.1)
Building wheels for collected packages: cocotb
  Building wheel for cocotb (PEP 517): started
  Building wheel for cocotb (PEP 517): finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: 'c:\python\python.exe' 'c:\python\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py' build_wheel 'C:\Users\Asus\AppData\Local\Temp\tmp98emznkb'
       cwd: C:\Users\Asus\AppData\Local\Temp\pip-install-1kuqem_c\cocotb_26c43314303f40e28b8563582f6ad2e1
  Complete output (93 lines):
  Modelsim/Questa executable (vdbg) executable not found. No FLI interface will be available.
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.9
  creating build\lib.win-amd64-3.9\cocotb
  copying cocotb\ANSI.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\binary.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\bus.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\clock.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\config.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\decorators.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\handle.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\ipython_support.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\log.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\memdebug.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\outcomes.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\queue.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\regression.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\result.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\scheduler.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\scoreboard.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\triggers.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\utils.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\wavedrom.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\xunit_reporter.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\_py_compat.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\_sim_versions.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\_version.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\__init__.py -> build\lib.win-amd64-3.9\cocotb
  creating build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\amba.py -> build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\avalon.py -> build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\opb.py -> build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\xgmii.py -> build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\__init__.py -> build\lib.win-amd64-3.9\cocotb\drivers
  creating build\lib.win-amd64-3.9\cocotb\generators
  copying cocotb\generators\bit.py -> build\lib.win-amd64-3.9\cocotb\generators
  copying cocotb\generators\byte.py -> build\lib.win-amd64-3.9\cocotb\generators
  copying cocotb\generators\packet.py -> build\lib.win-amd64-3.9\cocotb\generators
  copying cocotb\generators\__init__.py -> build\lib.win-amd64-3.9\cocotb\generators
[... 32 lines cut ...]
  copying cocotb\share\include\gpi_logging.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\py_gpi_logging.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\sv_vpi_user.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\vhpi_user.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\vhpi_user_ext.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\vpi_user.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\vpi_user_ext.h -> build\lib.win-amd64-3.9\cocotb\share\include
  creating build\lib.win-amd64-3.9\cocotb\share\def
  copying cocotb\share\def\.gitignore -> build\lib.win-amd64-3.9\cocotb\share\def
  copying cocotb\share\def\aldec.def -> build\lib.win-amd64-3.9\cocotb\share\def
  copying cocotb\share\def\ghdl.def -> build\lib.win-amd64-3.9\cocotb\share\def
  copying cocotb\share\def\icarus.def -> build\lib.win-amd64-3.9\cocotb\share\def
  copying cocotb\share\def\modelsim.def -> build\lib.win-amd64-3.9\cocotb\share\def
  copying cocotb\share\def\README.md -> build\lib.win-amd64-3.9\cocotb\share\def
  creating build\lib.win-amd64-3.9\cocotb\share\lib
  creating build\lib.win-amd64-3.9\cocotb\share\lib\verilator
[... cut ...]
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 21.1.1; however, version 21.1.2 is available.
You should consider upgrading via the 'c:\python\python.exe -m pip install --upgrade pip' command.

themperek, 2021-06-23:
**error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools"**

Or follow instructions: https://docs.cocotb.org/en/stable/install.html

themperek, 2021-06-23:
I will close this issue feel free to reopen it.

closed 2021-06-23 by themperek

Future-progrmer, 2021-06-23:
I have reinstalled cocotb and try to run a test from C:\cocotb-master\examples\adder\tests

>/usr/bin/tr: missing operand
Try '/usr/bin/tr --help' for more information.
c:\users\asus\miniconda3\lib\site-packages\cocotb\share\makefiles/Makefile.sim:108: *** "Couldn't find makefile for simulator: ""! Available simulators: vcs questa verilator ghdl icarus riviera activehdl xcelium ius modelsim cvc".  Stop.