cocotb/cocotb #2572: Expose actual type name of handles

ktbarrett, 2021-06-21:
Right now type names on handles are mapped to the GPI types which *removes* potentially useful information about the type from the handle. For example, I'd like to know if a `ModifiableObject` handle is actually an `sfixed` or a `std_logic_vector`. There is the possibility of overloading `get_type_str` on `GpiObject`s, but none of the implementations do this and just fallback to the default impl, which just serializes the GPI object type.

https://github.com/cocotb/cocotb/blob/59b6e9323517b9f7e5570029a6e3466d1d6a02b2/cocotb/share/lib/gpi/GpiCbHdl.cpp#L37-L64

marlonjames, 2021-06-22:
My thoughts on this are similar to #2195.
We should be careful with where and how we break the abstraction of GPI (if at all).

ktbarrett, 2021-06-22:
Having a wider interface in a subtype does not break Liskov substitution. If `sfixed` handles are still subtypes of `std_logic_vector` or `vpiNet` handles, and support everything `std_logic_vector`/`vpiNet` handles do, there is no invariant breakage. Additionally, the type string is not critical for operation and can be seen as orthogonal detail, meaning we could discount it from the "official" interface where we care about substitutability. If that's still not safe enough, we can support implicitly "downcasting" the implementation to one that supports type strings by doing a "has_type_string" check. This was what you proposed with the system tasks in #2195.

I understand this will break the current behavior of `simulator.get_type_string()` and the attribute in the handle objects that stores it; but simply serializing the GPI type enum (which is available at a different attribute already) is so useless I doubt anyone is depending on this functionality rn.