cocotb/cocotb #2573: cocotb verilator compilation fails with coverage enabled

sammy17, 2021-06-21:
I am getting the following Verilator compilation issue when I enable Verilator coverage with cocotb EXTRA_FLAGS env. 

How I enabled verilator coverage
`EXTRA_ARGS = --coverage`

Verilator compilation error when coverage is enabled:
```
Vtop__ALL.a(Vtop__123.o): In function `Vtop::_eval(Vtop__Syms*)':
Vtop__123.cpp:(.text+0x78c4): undefined reference to `Vtop::_sequent__TOP__120(Vtop__Syms*)'
Vtop__ALL.a(Vtop__108__Slow.o): In function `Vtop::_eval_settle(Vtop__Syms*)':
Vtop__108__Slow.cpp:(.text+0xdc22): undefined reference to `Vtop::_settle__TOP__2(Vtop__Syms*)'
collect2: error: ld returned 1 exit status
make[2]: *** [Vtop.mk:64: BoomTile] Error 1
```
I am using cocotb 1.5.2 with Verilator 4.106 2020-12-02 rev v4.106. 
Python version - 3.8.6

When I disable the coverage, the simulation proceeds fine without any errors. 

Please let me know if there is any way to resolve this issue.

ktbarrett, 2021-06-21:
Did you clean out the work directory before recompiling? I just tried this with `tests/test_cases/test_cocotb` and got compilation errors as well, but the errors I got pointed to an issue with Verilator's implementation of coverage.

```
Vtop.cpp: In static member function ‘static void Vtop::_combo__TOP__3(Vtop__Syms*)’:
Vtop.cpp:105:33: error: invalid operands of types ‘double’ and ‘double’ to binary ‘operator^’
  105 |     if ((vlTOPp->stream_in_real ^ vlTOPp->sample_module__DOT____Vtogcov__stream_in_real)) {
      |          ~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  |                        |
      |                  double                   double
Vtop.cpp:110:35: error: no match for ‘operator^’ (operand types are ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’} and ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’})
  110 |     if ((vlTOPp->stream_in_string ^ vlTOPp->sample_module__DOT____Vtogcov__stream_in_string)) {
      |          ~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  |                          |
      |                  basic_string<[...]>        basic_string<[...]>
```

Clearly there is no implementation of `^` for doubles or strings, and this is a bad assumption on Verilator's part. Perhaps this is resolved in newer version of Verilator? There are at least a few bugs in Verilator's VPI implementation that are preventing using a version newer than v4.106 with cocotb. Pinging @marlonjames.

sammy17, 2021-06-22:
@ktbarrett Thanks for the response. Yes, I cleaned the Verilator compilation directory completely and tried few times. Still getting an error similar to what I posted above with slight differences in the trace. 
**Error that I received in the latest compilation:**
```
Vtop__ALL.a(Vtop__163.o): In function `Vtop::_eval(Vtop__Syms*)':
Vtop__163.cpp:(.text+0x78c4): undefined reference to `Vtop::_sequent__TOP__117(Vtop__Syms*)'
Vtop__ALL.a(Vtop__107__Slow.o): In function `Vtop::_eval_initial(Vtop__Syms*)':
Vtop__107__Slow.cpp:(.text+0xb6): undefined reference to `Vtop::_initial__TOP__2(Vtop__Syms*)'
collect2: error: ld returned 1 exit status
make[2]: *** [Vtop.mk:64: BoomTile] Error 1
```

I tried enabling full coverage (--coverage option), only toggle coverage (--coverage-toggle option) and only line coverage (--coverage-line option) separately. Full coverage and toggle coverage both compilation failed with an undefined reference error as posted. Compilation with line coverage did not give any errors but it did not produce any **coverage.dat** file as well. 

I tried with newer Verilator versions (v4.110 and v4.204) but simulation gets stuck with these versions when coverage is not enabled. When coverage is enabled with v4.110, I get the same error that I got with the latest compile.

marlonjames, 2021-06-22:
I haven't used the coverage option with Verilator.
Are you able to simulate your design with coverage in a non-cocotb context?

sammy17, 2021-06-23:
No, I have not simulated this design without cocotb. RTL is quite complex and it will take some time to write a simulation testbench for this design for that purpose. But I will try to write a testbench to figure out this issue. 

I tried a much simpler design with cocotb and Verilator coverage enabled. That compilation and simulation went fine and generated a coverage.dat file as well. Therefore the issue seems to be triggered only in some scenarios. 

I checked the Verilator build directory for functions in the error (_sequent__TOP__117, _initial__TOP__2) and these function definitions are not there as the error suggests. 
So the issue seems to be in Verilator. Let me know if you have any suggestions to resolve this error. 
I will post this issue in Verilator github repo as well to see if it's encountered before after trying with the latest Verilator version.

sammy17, 2021-08-22:
I was able to get the latest Verilator (4.210) to work with cocotb with the fix mentioned in https://github.com/verilator/verilator/issues/2778#issuecomment-886149648.

Also, the issue of `coverage.dat `file not dumping was fixed with the edit I mentioned in https://github.com/cocotb/cocotb/issues/1478#issuecomment-903345076. 

I am closing this issue since the issues I have mentioned in this thread are now resolved with the above fixes.

closed 2021-08-22 by sammy17