cocotb/cocotb #2574: How to finish a test correctly?

Botnic, 2021-06-22:
I work with v1.6-dev and have the following issue:
When we generate a series of tests using TestFactory, they seem to run all in parallel. In the simulator one after the other is run, but in python they block the resources during the hole time. If to many tests are generated, I run out of threads.

>  File "/home/reto.meier/.pyenv/versions/3.9.1/lib/python3.9/threading.py", line 874, in start
> _start_new_thread(self._bootstrap, ())
>  RuntimeError: can't start new thread

The number of threads in Python is limited.
Is there a way to better close a test and free the resourced used by it (and release created threads)? At the moment each test finished by 
> raise tb.scoreboard.result

Thank you for your help
Kind regards
Reto Meier

eric-wieser, 2021-06-22:
Are you using threads within your tests themselves? Can you show the full traceback rather than just the last frame?

ktbarrett, 2021-06-22:
@Botnic Are you making heavy use of `cocotb.external` and `cocotb.function` decorators? They use threading resources in their implementation, and were only meant to be used sparingly.

Botnic, 2021-07-05:
Hi @eric-wieser 
Hi @ktbarrett

The issue was solved. It was a problem outside of cocotb.

closed 2021-07-05 by Botnic