cocotb/cocotb #2575: Deprecate TestFactory.generate_tests prefix and postfix

ktbarrett, 2021-06-22:
They originally existed so repeated usage of the same function with a `TestFactory` would not cause name conflicts. This was because previously the generated test name was simply the test name with a counter appended to it. Now that `TestFactory` generated tests use the option values in the test name, conflict can only occur if the same test with the same options occur in different `TestFactory` calls. This should instead cause a warning to be printed and the conflict resolved by appending a counter to the test name of the conflicting test.

jahagirdar, 2023-01-19:
Is this still open for discussion? The proposed deprecation will impact testcase reporting for my usecase.

The way I write tests is:

I have one single python test file for each environment which performs the following actions.
```
cfg=load_cfg(os.path.join(os.getenv('TESTDIR'),os.getenv('TESTCASE'))
env=ENV(dut,cfg)
env.setup()
await env.reset()
await env.initialize()
env.start()
await env.run()
while ! test_done:
 await Timer(20,'us')
env.stop()

factory = TestFactory(run__test)
factory.generate_tests(prefix=os.getenv('TESTDIR').replace(os.getenv('VERIF_ROOT'), 'DYU') + "__", postfix="__" + os.getenv('TESTCASE_NAME') + "__")
```
I have hundreds of tests which use the same python test file and use test specific configuration file to configure and run different scenarios.
Without the prefix and suffix options my regression report will only contain entries for `run_test` instead of the current `testsuitename_run_test_testcasename`

marlonjames, 2023-01-19:
It is still open.

This issue is linked to the changes discussed #2548.

A change was made and then reverted, so for now the prefix/postfix remains.
- #2566 
- #2617

ktbarrett, 2023-01-19:
@jahagirdar I don't think your use case was the intent of this feature and is what I'd consider an abuse of a feature. From what I can tell it was meant to prevent collisions using the same test with multiple TestFactory calls. That being said I don't think this is up for debate as long as the TestFactory and the whole RegressionManager remain implemented as they currently do, which I anticipate they will for a while. I'll close this and if something changes in the future a new issue can be opened.

closed 2023-01-19 by ktbarrett