cocotb/cocotb #2582: Add remote pdb support to interrupt hanging process

ktbarrett, 2021-06-27:
Many times when debugging new Components I end up in a position where the simulation is running, but nothing is happening. In those cases it would be nice to break into a debugger and inspect the state of the test to find out why it's hung. This could be done by [registering a signal handler that would break into pdb](http://blog.devork.be/2009/07/how-to-bring-running-python-program.html), or [remote pdb](https://github.com/ionelmc/python-remote-pdb) for simulators that capture stdin, like Questa.

eric-wieser, 2021-06-27:
See also https://pypi.org/project/pdb-attach/, which would probably make this very easy to do today, assuming we don't get screwed by the embedded debugger.

ktbarrett, 2021-06-27:
Just tested this out with the matrix_multiplier example by adding the `pdb_attach.listen()` call to the top of the test and it works. I'll consider this feature out-of-scope, but also working and tested.

eric-wieser, 2021-06-27:
I think the use case is common enough here that it might be worth mentioning `pdb_attach` in our documentation.

ktbarrett, 2021-06-27:
Some basic debugging tutorials are appropriate. This can definitely be mentioned in them.

ktbarrett, 2024-12-22:
The docs now have documentation on using `remote_pdb`.

closed 2024-12-22 by ktbarrett