cocotb/cocotb #2594: GHDL unable to access signals in generate loops

ktbarrett, 2021-06-28:
## Description

Unable to access scopes or any object within a scope in a generate loop using GHDL's VPI.

```
     0.00ns DEBUG    VPI: Created GPI object from type ???(32)
     0.00ns DEBUG    Found asc_gen via VPI
     0.00ns DEBUG    Checking array_module.asc_gen exists
     0.00ns DEBUG    Created
     0.00ns DEBUG    Checking if index 20 native through implementation VPI
     0.00ns DEBUG    Native check create for index 20 of parent asc_gen (pseudo-region)
 vpi_handle_by_name ("array_module.asc_gen[20]", 0000000000000000) = 0000000000000000
     0.00ns DEBUG    Unable to vpi_get_handle_by_index asc_gen[20]
```

eefresher, 2021-08-20:
[image]

i meet the same problem with you ,use the `_id()`, can solve it

ktbarrett, 2021-08-21:
It does not have to do with underscores. It cannot find `dut.asc_gen[20]`.

It can find `dut.asc_gen` pseudo-region. After looking a bit harder I noticed that it seems like the issue is that GHDL uses different array indexing syntax. It finds `dut.asc_gen(16)`.

```
     0.00ns INFO     running test_gen_loop (1/1)
     0.00ns DEBUG    Searching for asc_gen
     0.00ns DEBUG    Checking if asc_gen is native through implementation VPI
     0.00ns DEBUG    Found pseudo-region array_module.asc_gen (hdl_name=asc_gen(16) but name=asc_gen)
vpi_get_str: unhandled property
vpi_get_str: unhandled property
     0.00ns DEBUG    VPI: Created GPI object from type ???(32)
     0.00ns DEBUG    Found asc_gen via VPI
     0.00ns DEBUG    Checking array_module.asc_gen exists
     0.00ns DEBUG    Created
     0.00ns DEBUG    Checking if index 20 native through implementation VPI 
     0.00ns DEBUG    Native check create for index 20 of parent asc_gen (pseudo-region)
     0.00ns DEBUG    Unable to vpi_get_handle_by_index asc_gen[20]
     0.00ns WARNING  Failed to find a handle at index 20 via any registered implementation
     0.00ns DEBUG    _initialise_testbench successful
     0.00ns INFO     test_gen_loop passed: errored as expected (result was IndexError)
```

eefresher, 2021-08-21:
` _id()` is also useful for get handl of signal in generate loop.  for your problem, you can try `dut._id("xxxx",extended=false)`,in which `xxx` is the signal hierarchy .