cocotb/cocotb #2605: Questa VPI unable to access logic array if last dimension is unpacked

ktbarrett, 2021-07-01:
## Description

Using Questa's VPI, we are unable to access into the first dimension of a logic array if and only if the last dimension is unpacked. Mixed packed-ness custom array types work just fine. Apparently, this is fixed by some of the changes in #826. But I'm not sure which set of changes fix the issue.

```systemverilog
  input     logic               [2:0][2:0]          in_vect_packed_packed_unpacked[2:0];
```

```
# Setting: dut.in_vect_packed_packed_unpacked type <class 'cocotb.handle.NonHierarchyIndexableObject'>
#      1.00ns DEBUG    in_vect_packed_packed_unpacked has 3 elements
#      1.00ns DEBUG    Checking if index 2 native through implementation VPI 
#      1.00ns DEBUG    Unable to find handle through vpi_handle_by_index(), attempting second method
#      1.00ns DEBUG    Unable to vpi_get_handle_by_index in_vect_packed_packed_unpacked[2]
#      1.00ns WARNING  Failed to find a handle at index 2 via any registered implementation
```

## Failing Tests

* `/tests/test_cases/test_cocotb_array/test_cocotb_array:test_in_vect_packed_packed_unpacked`
* `/tests/test_cases/test_cocotb_array/test_cocotb_array:test_in_vect_packed_unpacked`
* `/tests/test_cases/test_cocotb_array/test_cocotb_array:test_in_vect_unpacked`

raysalemi, 2021-07-16:
Has this been raised through FOSSI?  Is this a Questa bug?

ktbarrett, 2021-07-16:
@raysalemi #826 fixed the problem, but the patch was rejected because it wasn't up to quality standards. I'm not sure if it fixed the problem by working around an underlying Questa bug, or fixed a cocotb bug. I will investigate this at some point an mark this issue with the `upstream` tag and open a ticket through FOSSi if it is a Questa bug.

raysalemi, 2021-07-16:
Perfect!

I’m glad we’ve got the FOSSI partnership now.