cocotb/cocotb #2608: PEP517 Error while installing cocotb package

Nimrod123, 2021-07-03:
Building wheel for cocotb (PEP 517) ... error
ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly

I tried updating python and other packages. Also I tried to install the development version, the same error seems to pop up.
I have installed python 3.9 version on Windows 10.

ktbarrett, 2021-07-03:
Could you post the full error?

Nimrod123, 2021-07-03:
`pip install cocotb`
Collecting cocotb
  Using cached cocotb-1.5.2.tar.gz (579 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting cocotb-bus<1.0
  Using cached cocotb_bus-0.1.1-py3-none-any.whl (34 kB)
Building wheels for collected packages: cocotb
  Building wheel for cocotb (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\users\nisha\appdata\local\programs\python\python39\python.exe' 'c:\users\nisha\appdata\local\programs\python\python39\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py' build_wheel 'C:\Users\nisha\AppData\Local\Temp\tmpt7ujk8bl'
       cwd: C:\Users\nisha\AppData\Local\Temp\pip-install-grr213xt\cocotb_667dba73e38a4985bd5ff954c22aeb03
  Complete output (92 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.9
  creating build\lib.win-amd64-3.9\cocotb
  copying cocotb\ANSI.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\binary.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\bus.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\clock.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\config.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\decorators.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\handle.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\ipython_support.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\log.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\memdebug.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\outcomes.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\queue.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\regression.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\result.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\scheduler.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\scoreboard.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\triggers.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\utils.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\wavedrom.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\xunit_reporter.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\_py_compat.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\_sim_versions.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\_version.py -> build\lib.win-amd64-3.9\cocotb
  copying cocotb\__init__.py -> build\lib.win-amd64-3.9\cocotb
  creating build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\amba.py -> build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\avalon.py -> build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\opb.py -> build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\xgmii.py -> build\lib.win-amd64-3.9\cocotb\drivers
  copying cocotb\drivers\__init__.py -> build\lib.win-amd64-3.9\cocotb\drivers
  creating build\lib.win-amd64-3.9\cocotb\generators
  copying cocotb\generators\bit.py -> build\lib.win-amd64-3.9\cocotb\generators
  copying cocotb\generators\byte.py -> build\lib.win-amd64-3.9\cocotb\generators
  copying cocotb\generators\packet.py -> build\lib.win-amd64-3.9\cocotb\generators
  copying cocotb\generators\__init__.py -> build\lib.win-amd64-3.9\cocotb\generators
  creating build\lib.win-amd64-3.9\cocotb\monitors
  copying cocotb\monitors\avalon.py -> build\lib.win-amd64-3.9\cocotb\monitors
  copying cocotb\monitors\xgmii.py -> build\lib.win-amd64-3.9\cocotb\monitors
  copying cocotb\monitors\__init__.py -> build\lib.win-amd64-3.9\cocotb\monitors
  creating build\lib.win-amd64-3.9\cocotb\_vendor
  copying cocotb\_vendor\__init__.py -> build\lib.win-amd64-3.9\cocotb\_vendor
  creating build\lib.win-amd64-3.9\cocotb\_vendor\find_libpython
  copying cocotb\_vendor\find_libpython\_version.py -> build\lib.win-amd64-3.9\cocotb\_vendor\find_libpython
  copying cocotb\_vendor\find_libpython\__init__.py -> build\lib.win-amd64-3.9\cocotb\_vendor\find_libpython
  copying cocotb\_vendor\find_libpython\__main__.py -> build\lib.win-amd64-3.9\cocotb\_vendor\find_libpython
  creating build\lib.win-amd64-3.9\cocotb\share
  creating build\lib.win-amd64-3.9\cocotb\share\makefiles
  copying cocotb\share\makefiles\Makefile.deprecations -> build\lib.win-amd64-3.9\cocotb\share\makefiles
  copying cocotb\share\makefiles\Makefile.inc -> build\lib.win-amd64-3.9\cocotb\share\makefiles
  copying cocotb\share\makefiles\Makefile.sim -> build\lib.win-amd64-3.9\cocotb\share\makefiles
[... 15 lines cut ...]
  copying cocotb\share\include\exports.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\gpi.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\gpi_logging.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\py_gpi_logging.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\sv_vpi_user.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\vhpi_user.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\vhpi_user_ext.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\vpi_user.h -> build\lib.win-amd64-3.9\cocotb\share\include
  copying cocotb\share\include\vpi_user_ext.h -> build\lib.win-amd64-3.9\cocotb\share\include
  creating build\lib.win-amd64-3.9\cocotb\share\def
  copying cocotb\share\def\.gitignore -> build\lib.win-amd64-3.9\cocotb\share\def
[... cut ...]
  creating build\lib.win-amd64-3.9\cocotb\share\lib\verilator
  copying cocotb\share\lib\verilator\verilator.cpp -> build\lib.win-amd64-3.9\cocotb\share\lib\verilator
  running build_ext
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/        
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly

themperek, 2021-07-03:
`error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/`

You can use GCC need to follow instructions in documentation or msys2 or install visual studio.

Nimrod123, 2021-07-04:
I installed the visual C++14.29  and tried to install cocotb again. There seems to be some missing lib or obj files related to the linker

Creating library build\lib.win32-3.9\cocotb\libs\pygpilog.lib and object build\lib.win32-3.9\cocotb\libs\pygpilog.exp
  py_gpi_logging.obj : error LNK2001: unresolved external symbol __imp__PyLong_FromLong
  py_gpi_logging.obj : error LNK2001: unresolved external symbol __imp__PyGILState_Release
  py_gpi_logging.obj : error LNK2001: unresolved external symbol __imp___Py_Dealloc
  py_gpi_logging.obj : error LNK2001: unresolved external symbol __imp__PyObject_CallFunctionObjArgs
  py_gpi_logging.obj : error LNK2001: unresolved external symbol __imp__PyGILState_Ensure
  py_gpi_logging.obj : error LNK2001: unresolved external symbol __imp__PyObject_IsTrue
  py_gpi_logging.obj : error LNK2001: unresolved external symbol __imp__PyErr_Print
  py_gpi_logging.obj : error LNK2001: unresolved external symbol __imp__PyUnicode_FromString
  build\lib.win32-3.9\cocotb\libs\pygpilog.dll : fatal error LNK1120: 8 unresolved externals
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30037\\bin\\HostX86\\x86\\link.exe' failed with exit code 1120
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly

ktbarrett, 2021-07-04:
> build\lib.win32-3.9

You've installed a 32-bit version of MSVC, so you'll need a 32-bit distribution of Python. Are you on a 32-bit system?

Nimrod123, 2021-07-04:
I've a 64-bit system. I reinstalled MSVC and now it works. 

Thanks for your time.

closed 2021-07-04 by Nimrod123