cocotb/cocotb #2611: Make error

Nimrod123, 2021-07-04:
$ make SIM=modelsim 
C:/MinGW/bin/make results.xml
make[1]: Entering directory 'C:/Users/nisha/OneDrive/Documents/GitHub/NCGPythonCocotb/tb'
make[1]: *** No rule to make target '/c/Users/nisha/OneDrive/Documents/GitHub/NCGPythonCocotb/tb/sumofN.v', needed by 'sim_build/runsim.do'.  Stop.
make[1]: Leaving directory 'C:/Users/nisha/OneDrive/Documents/GitHub/NCGPythonCocotb/tb'
c:\users\nisha\appdata\local\programs\python\python39\lib\site-packages\cocotb\share\makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2

I have modelsim installed.

ktbarrett, 2021-07-04:
> ```
> make[1]: *** No rule to make target '/c/Users/nisha/OneDrive/Documents/GitHub/NCGPythonCocotb/tb/sumofN.v', needed by 'sim_build/runsim.do'.  Stop.

Looks like it doesn't see that file. Is the file there? Name misspelled? Path slightly wrong?

Nimrod123, 2021-07-04:
```
# Makefile
SIM ?= modelsim
TOPLEVEL_LANG ?= verilog
VERILOG_SOURCES = $(shell pwd)/sumofN.v
TOPLEVEL := sumofN
MODULE := SumN_tb

include $(shell cocotb-config --makefiles)/Makefile.sim
```

I have included the .py,   .v and makefile in the same folder I opened the shell. File names and module names seems to be correct but it seems as if it cant find such file

ktbarrett, 2021-07-04:
And you are sure you can access that file using that same filepath in the error message? The reason I ask is because I see `OneDrive` in the path. I don't know how well network drives are supported by mingw.

Are you using the free version of ModelSim? Because that version is 32 bit only,  soyou'll need a 32 bit Python and 32 bit compiler.

Nimrod123, 2021-07-05:
`C:/MinGW/bin/make results.xml
make[1]: Entering directory 'C:/Users/nisha/Music/tb'
make[1]: *** No rule to make target '/c/Users/nisha/Music/tb/sumofN.v', needed by 'sim_build/sim.vvp'.  Stop.
make[1]: Leaving directory 'C:/Users/nisha/Music/tb'
c:\users\nisha\appdata\local\programs\python\python39\lib\site-packages\cocotb\share\makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2`

I was using the 32 bit version of modelsim. Now, I installed icarus verilog, checked as 64 bit version.  I tried again but I had the same issue of `no rule to make target`. I changed the files to some local path and tried it, still same error pops up.

I am using a 64 bit system, 64 bit version python and compiler.

closed 2021-07-07 by Nimrod123