cocotb/cocotb #2622: Can't pass the examples of mixed_langauge using Questa

jbk4312, 2021-07-15:
# Loading sv_std.std
# Loading work.endian_swapper_mixed(fast)
# Loading work.endian_swapper_sv(fast)
# Loading std.standard
# Loading std.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading ieee.numeric_std(body)
# Loading work.endian_swapper_vhdl(impl)#1
# Loading C:/python39/lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll
#      -.--ns ERROR    cocotb.gpi                         ..s\cocotb_utils.cpp:57   in utils_dyn_open                  **Unable to open lib C;C:** The specified module could not be found.
#
#
# cocotb: Error loading shared library C;C
# End time: 17:14:21 on Jul 15,2021, Elapsed time: 0:00:09
# Errors: 0, Warnings: 1
ERROR: results.xml was not written by the simulation!
/c/python39/lib/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:145: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/c/Users/****/git/cocotb/examples/mixed_language/tests'
c:\python39\lib\site-packages\cocotb\share\makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2

I am using the conda environments in windows 10 and then failed the testing of examples

ktbarrett, 2021-07-15:
>  **Unable to open lib C;C:** 

That doesn't look right. Does your path contain non-ASCII characters or is it *really* long? I don't think we are handling non-ASCII characters correctly right now.

jbk4312, 2021-07-15:
Miniconda virtual environments -> (coco)

(coco) c:\Users\jbk43\mixed_language\tests>make SIM=questa

I tried with a little shorter path and then can't succeed also. All other examples except mixed_signal with Questa succeeded. The sim_build directory was created but __pycache__ wasn't created so it doesn't compile test_mixed_language.py for whaterver reasons.

ktbarrett, 2021-07-15:
> All other examples except mixed_signal with Questa succeeded
> test_mixed_language.py

Is the issue mixed_signal, mixed_language, endian_swapper, or everything?

imphil, 2021-07-15:
I saw the same error message here: https://github.com/cocotb/cocotb/discussions/2529#discussioncomment-1001834 (second screenshot). Which parts of the environment described in this other issue match yours, @jbk4312? Maybe we can narrow it down a bit.

jbk4312, 2021-07-15:
Questa doesn't support the mixed_signal but it worked for analog something case in the example. And underscore underscore pycache underscore underscore directory must be created below the /examples/tests/__pycache__ directory but I can't find it so the problems may be related to pre compiling users' python file. It was same to other issues already reported at #2529.

cmarqu, 2021-07-15:
Maybe it's missing FLI but we are not getting the error message right (which got introduced with https://github.com/cocotb/cocotb/commit/94064f9d2018e835eac27b65619fafef0a94d3d8)?

ktbarrett, 2021-07-15:
We are using the ASCII-suffixed functions and not unicode.

jbk4312, 2021-07-16:
# Autogenerated file
onerror {
	quit -f -code 1
}
if [file exists sim_build/work] {vdel -lib sim_build/work -all}
vlib sim_build/work
vmap -c
vmap work sim_build/work
vcom -work work +acc C:/Users/jbk43/git/cocotb/examples/mixed_language/hdl/endian_swapper.vhdl
vlog -work work +define+COCOTB_SIM -sv -timescale 1ns/1ps -mfcu +acc  C:/Users/jbk43/git/cocotb/examples/mixed_language/hdl/endian_swapper.sv C:/Users/jbk43/git/cocotb/examples/mixed_language/hdl/top.sv

vsim **-pli C:/python39/lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll** -onfinish exit **-pli C:/python39/lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll**   sim_build/work.endian_swapper_mixed

onbreak resume
run -all
quit 

Probably I found the problems that caused this. It came from vsim command line entering double times this argument of "-pli C:/python39/lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll". Maybe the simulator misunderstood this argument needs twice because there are verilog & vhdl mixed sources. But I don't know how to fix it.

closed 2021-07-16 by jbk4312