cocotb/cocotb #2625: Launching coroutines using cocotb

raysalemi, 2021-07-18:
Does cocotb have a the equivalent of `asyncio.run()`?  I'd like to test cocotb-based coroutines without a simulator.

ktbarrett, 2021-07-18:
This does not exist (yet). I'd like to see this feature supported by the scheduler rewrite that @marlonjames is working on. Do you expect triggers to be supported, or only coroutines?

raysalemi, 2021-07-18:
I've extended `cocotb.queues.Queue` to add peek functionality (for TLM). I'd like to test it in my current `nosetest` suite.

```
    async def data_transfer(self):
        await self.put_data(self. data_list)
        await self.get_data()

    def test_data_transfer(self):
        self.data_list = [1,"2", "three", None]
        Scheduler.start_soon(self.data_transfer())
        self.assertEqual(self.data_list, self.got_data)
```

So it seem to me that I'd need a way to run a coroutine and wait for it to return. I'm not sure if that requires triggers.

raysalemi, 2021-07-18:
I'll be building from here to test `tlm_fifo`, etc.

eric-wieser, 2021-07-18:
What do you mean by `TLM`?

ktbarrett, 2021-07-18:
@raysalemi 
> I'm not sure if that requires triggers

It requires `Event`, but not *simulator* triggers for timing, value change, etc. There are others interested in running without a simulator, but they want to mock out the entire simulator so they can, for example, unit test Drivers and Monitors. This is a bit different of a request and easier to accomplish.

@eric-wieser "Transaction Level Modeling". There is a standard describing it. TLM queues are the standard message passage queue for UVM.

raysalemi, 2021-07-18:
I've now got testing running by using the @cocotb.test() decorator.

ktbarrett, 2021-07-18:
I'm going to keep this issue open so we can track it.

PiyushSaini97, 2024-11-07:
Wanted to know if this feature will be implemented?

ktbarrett, 2024-11-07:
@PiyushSaini97 This feature in particular is pretty low priority, but it might become very easy to implement after a coming refactor. Another possible solution is to use asyncio in place of cocotb when not using a simulator. We've put effort into trying to align the cocotb and asyncio APIs to allow this to happen.

A simple compat layer could do. Here's some initial code to that end. I know you already have a wrapper for `cocotb.test`.
```python
if cocotb.is_simulation:
    import cocotb.task, cocotb.triggers
    Task = cocotb.task.Task
    Event = cocotb.triggers.Event
    Lock = cocotb.triggers.Lock
    start_soon = cocotb.start_soon
else:
   import asyncio
   Task = asyncio.Task
    Event = asyncio.Event
    Lock = asyncio.Lock
    start_soon = asyncio.create_task
```