cocotb/cocotb #2626: Enhancement idea

raysalemi, 2021-07-18:
Would it make sense to add `peek` and `peek_nowait` to the cocotb.queues.Queue`?

eric-wieser, 2021-07-18:
Can you give an example of an application where `peek` would be useful? I assume you're thinking of a single-consumer queue, since `peek` is useless on a multi-consumer queue?

Note some other Asyncio libraries have rejected this kind of feature request, eg https://github.com/aio-libs/janus/issues/67

eric-wieser, 2021-07-18:
It's quite straightforward to build what you're asking for as a standalone component
```python
class InspectableQueueConsumer:
    """ This object is not safe to use from multiple coroutines """
    def __init__(self, queue):
        self.queue = queue
        self._peeked = []

    async def get(self):
        try:
            return self._peeked.pop(0)
        except Indexerror:
            return await self.queue.get()

    async def peek(self):
        if not self.peeked:
            self.peeked.append(await self.queue.get())
        return self.peeked[0]
```

raysalemi, 2021-07-18:
> Can you give an example of an application where `peek` would be useful? I assume you're thinking of a single-consumer queue, since `peek` is somewhere between useless and dangerous on a multi-consumer queue?
> 
> Note some other python libraries and even python itself have rejected this kind of feature request, eg [aio-libs/janus#67](https://github.com/aio-libs/janus/issues/67) and https://groups.google.com/g/comp.lang.python/c/ujy2LIHUy3o and https://bugs.python.org/issue35118

I'm modifying `pyuvm` to use coroutines instead of threads as @ktbarrett suggested.  The UVM TLM FIFO has a peek function and I have extended Queue to provide it to myself (as you suggested).  

I must admit, I have never used peek in anger, but I'm following the spec.

raysalemi, 2021-07-18:
> It's quite straightforward to build what you're asking for as a standalone component
>
> ```python
> class InspectableQueueConsumer:
>     """ This object is not safe to use from multiple coroutines [...]

I just modified the `get` pattern in the current code:

```python
 def _peek(self):
        return self._queue[0]

    async def peek(self):
        """Remove and return an item from the queue.
        If the queue is empty, wait until an item is available.
        """
        while self.empty():
            event = Event('{} get'.format(type(self).__name__))
            self._getters.append((event, cocotb.scheduler._current_task))
            await event.wait()
        return self.peek_nowait()

    def peek_nowait(self):
        """Remove and return an item from the queue.
        Return an item if one is immediately available, else raise
        :exc:`asyncio.QueueEmpty`.
        """
        if self.empty():
            raise QueueEmpty()
        item = self._get()
        self._wakeup_next(self._putters)
        return item
```

eric-wieser, 2021-07-18:
Your `peek_nowait` operation has a critical bug, as it consumes the data from the underlying queue making it no different to `get_nowait`.

ktbarrett, 2021-07-18:
`peek` and `peek_nowait` sound like a good idea.

raysalemi, 2021-07-18:
> Your `peek_nowait` operation has a critical bug, as it consumes the data from the underlying queue making it no different to `get_nowait`.

Funny. I just read this after I discovered the bug in my testing.