cocotb/cocotb #2628: asyncio.queues.QueueFull?

raysalemi, 2021-07-18:
Would it make sense to create cocotb.queues.QueueFull by extending asyncio.queues.QueueFull so that the user would not be tempted to import `asyncio`?

eric-wieser, 2021-07-18:
This seems like a good idea to me. I'm not even convinced that we should extend the asyncio exceptions at all.

@ktbarrett, do you feel strongly about extending the asyncio exceptions vs defining our own? The asyncio defs are just
```
class QueueFull(Exception):
    """Raised when the Queue.put_nowait() method is called on a full Queue."""
    pass
```
so we're not really gaining anything by extending them.

ktbarrett, 2021-07-18:
I don't really see a reason to extend it. They are available to be imported from `cocotb.queue` already, maybe we just need to document that?

eric-wieser, 2021-07-18:
The downside is that when you write `cocotb.queue.QueueEmpty` in a console, you get a `repr` containing `asyncio`, which is misleading because nothing else from asyncio will actually work from cocotb.

ktbarrett, 2021-07-18:
If that's a problem, then we to extend them. It's just a couple lines of code. @raysalemi Could you make a PR with this and the `peek` enhancement (#2626)?

```python
class QueueFull(asyncio.queues.QueueFull):
    """Raised when the Queue.put_nowait() method is called on a full Queue."""

class QueueEmpty(asyncio.queues.QueueEmpty):
    """Raised when the Queue.get_nowait() method is called on a empty Queue."""
```

eric-wieser, 2021-07-18:
I think we should do this in two PRs, simply because there's a lot of precedent against a `peek` API being pythonic, but the exceptions sound uncontroversial.

raysalemi, 2021-07-18:
Sorry. I just saw this. I have one PR for both.  Want me to resubmit?

ktbarrett, 2021-07-20:
Resolved by #2631.

closed 2021-07-20 by ktbarrett