cocotb/cocotb #2632: Documentation Vagueness on Environment Variables

raysalemi, 2021-07-19:
The [building documentation page](https://docs.cocotb.org/en/stable/building.html) should clearly differentiate between environment variables and makefile variables.

The subheading below named **Cocotb** should be named **Environment Variables**
---
# Variables

The following sections document environment variables and makefile variables according to their owner/consumer.

Of the environment variables, only MODULE is mandatory to be set (typically done in a makefile or run script), all others are optional.

## Cocotb
---

eric-wieser, 2021-07-19:
Note that the source does actually distinguish them, although I'm not certain they are correctly partitioned.

https://github.com/cocotb/cocotb/blob/master/documentation/source/building.rst

I think this is a mess because of how environment variables appear as make variables and to a lesser extent vice versa. Perhaps we should remind the reader of that duality before listing anything.

raysalemi, 2021-07-19:
The feature to limit messages only works as an environment variable.

ktbarrett, 2021-07-19:
It should be noted that this vagueness goes away once we replace the Makefiles with the Python runner.

eric-wieser, 2021-07-19:
Replacing the makefiles doesn't help people who want want use cocotb in a custom flow that uses their own makefiles / integrates with some other software etc.

Did we end up writing documentation for how to use the GPI dlls directly? I remember an issue for that at some point.

cmarqu, 2021-07-21:
@eric-wieser Are you thinking of https://docs.cocotb.org/en/stable/custom_flows.html ?