cocotb/cocotb #2633: Xcelium VHDL libraries not order preserving

martijnbastiaan, 2021-07-19:
In https://github.com/cocotb/cocotb/pull/2614 I added VHDL library support for Xcelium. Unfortunately, the more libraries a project has, the more likely it is that the introduced logic breaks. This is an interaction of two facts:

1. `-makelib` is order dependent (unlike `ghdl -i`). That is, in order for a library to compile all its dependencies should precede it.
2. Makefile's `$.VARIABLES` is not ordered non-deterministicly. 

For (2) consider:

```Makefile
FOO_1 = a
FOO_2 = b
FOO_3 = c
FOO_4 = d
FOO_5 = e
FOO_6 = f
FOO_7 = g

all:
	@echo "$(.VARIABLES)"
```

running it yields:

```
$ make | tr ' ' '\n' | grep -e 'FOO_[0-9]'
FOO_5
FOO_1
FOO_7
FOO_2
FOO_4
FOO_3
FOO_6
```

I'm unsure how to tackle this one. Perhaps introduce a `VHDL_LIBRARY_ORDER`? In general, I don't think there is a way to automate its population, although most users could probably do:

```Makefile
VHDL_LIBRARY_ORDER = $(shell grep -o -e '^VHDL_SOURCES_[^ ]*' Makefile  | tr '\n' ' ')
```

cmarqu, 2021-07-19:
xrun has a `-smartorder` switch, but I'm not sure if it only works for files in a single library or across libraries too?

martijnbastiaan, 2021-07-19:
I've just tested that, that seems to only work for individual files.

martijnbastiaan, 2021-07-19:
I've made a proof-of-concept for `VHDL_LIB_ORDER`. I couldn't make it properly work in the Makefile language, but since we're running cocotb we can be pretty sure that Python is installed, so:

```Makefile
###################
# User's Makefile #
###################
VHDL_SOURCES_a = x
VHDL_SOURCES_b = y

VHDL_LIB_ORDER = a c  

#######################
# In Makefile.xcelium #
#######################
define SCRIPT
import sys

def parse_variables(vars):
    return set(vars.split())

VHDL_SOURCES_VARS = "$(filter VHDL_SOURCES_%, $(.VARIABLES))"             
VHDL_SOURCES_VARS = {s[len("VHDL_SOURCES_"):] for s in parse_variables(VHDL_SOURCES_VARS)}
                                                      
VHDL_LIB_ORDER = "$(VHDL_LIB_ORDER)"                  
VHDL_LIB_ORDER = parse_variables(VHDL_LIB_ORDER)
   
if not VHDL_SOURCES_VARS:
    sys.exit(0)
    
if VHDL_SOURCES_VARS == VHDL_LIB_ORDER:
    sys.exit(0)
                                                      
if VHDL_SOURCES_VARS and not VHDL_LIB_ORDER:        
    print("VHDL_LIB_ORDER is a mandatory variable when using VHDL_SOURCES_* and simulating with Xcelium.")
    sys.exit(1)          
    
too_many = VHDL_LIB_ORDER - VHDL_SOURCES_VARS
if too_many:
    forgotton_vars = ["VHDL_SOURCES_" + l for l in too_many]        
    unexpected_err = "Unexpected libraries in VHDL_LIB_ORDER: {}.".format(", ".join(too_many))  
    hint_err = "Did you forget to define {}?".format(", ".join(forgotton_vars))
    print("ERROR: {} {}".format(unexpected_err, hint_err), file=sys.stderr)
                   
too_few = VHDL_SOURCES_VARS - VHDL_LIB_ORDER 
if too_few: 
    print("ERROR: Missing libraries in VHDL_LIB_ORDER: {}.".format(", ".join(too_few)), file=sys.stderr)                            
    
sys.exit(1)    

                                              
endef                                         
                                              
export SCRIPT                           

all:
        @echo "$$SCRIPT" > script.py   
        @python3 script.py             
        @rm script.py

```

The script logic tries to be helpful by suggesting what the user could do to solve the problem. E.g. for the given example it would:

```
$ make 
ERROR: Unexpected libraries in VHDL_LIB_ORDER: c. Did you forget to define VHDL_SOURCES_c?
ERROR: Missing libraries in VHDL_LIB_ORDER: b.
make: *** [Makefile:49: all] Error 1
```

Of course, the script would be written to `sim_build` instead in the real version. Finally, `Mkaefile.xcelium` could be simplified as:

```diff
diff --git a/cocotb/share/makefiles/simulators/Makefile.xcelium b/cocotb/share/makefiles/simulators/Makefile.xcelium
index d3040be3..7c26094c 100644
--- a/cocotb/share/makefiles/simulators/Makefile.xcelium
+++ b/cocotb/share/makefiles/simulators/Makefile.xcelium
@@ -103,13 +103,8 @@ else
     $(error "A valid value (verilog or vhdl) was not provided for TOPLEVEL_LANG=$(TOPLEVEL_LANG)")
 endif
 
-# Returns arguments to compile single VHDL library
-define add_lib
-       -makelib $(SOURCES_VAR:VHDL_SOURCES_%=%) $($(SOURCES_VAR)) -endlib
-endef
-
 # Builds a list of arguments to support VHDL libraries specified in VHDL_SOURCES_*:
-LIBS := $(foreach SOURCES_VAR, $(filter VHDL_SOURCES_%, $(.VARIABLES)),$(add_lib))
+LIBS := $(foreach LIB, $(VHDL_LIB_ORDER),-makelib $(LIB) $(VHDL_SOURCES_$(LIB)) -endlib)
 
 $(COCOTB_RESULTS_FILE): $(HDL_SOURCES) $(CUSTOM_COMPILE_DEPS) $(CUSTOM_SIM_DEPS) | $(SIM_BUILD)
        -@rm -f $(COCOTB_RESULTS_FILE)
```

Would you be receptive to such a patch?

ktbarrett, 2021-07-19:
Building a Python script to do work that Makefiles can't handle? Seems like we shouldn't bother and instead ensure that the Python runner supports building VHDL libraries correctly in the given order. We decided to replace the makefiles with cocotb-test by default before the next release. cocotb-test doesn't support specifying VHDL libraries but I made a few suggestions to @themperek on how I thought that should look. #1562

martijnbastiaan, 2021-07-19:
That's correct @ktbarrett and I agree it's a non-ideal. It really does show why a Python based runner is something we desperately need, and I'm happy to read that it's going to happen!

I'll admit that it puts me in a bit of an akward spot for two reasons:

1. I submitted - for all intents and purposes - broken code into cocotb (if only I had tested with more than 2 libraries..)
2. The client I'm currently contracting for needs Xcelium VHDL library support to work with cocotb for their current project

I don't feel happy about (1) for hopefully obvious reasons. For (2), I'd like to leave my client without a private fork of cocotb they'd need to maintain (especially because they're using Subversion internally, so maintaining it would mean a bit of effort). I realise my short-terms goals cross with cocotbs long-term (medium-term?) goals somewhat, but I'd like to plead for inclusion of a patch based on the PoC I posted. (Perhaps stating the obvious, but I'll work on getting it in a mergeable state. I don't expect implementation effort from the cocotb team.)

ktbarrett, 2021-07-19:
I'm surprised that a people are using the HEAD of cocotb. The makefiles will live on even after we replace them, so I would like this feature to work, or to be removed before the next release. This seems like a convoluted fix, is there really no way to loop over the `VHDL_LIB_ORDER` variable, compute the `VHDL_SOURCES_{lib}` variable name and get the value in Makefile's scripting language? Have you thought about moving your client to `cocotb-test` and pushing the missing feature there? This fix seems a little convoluted, but I'm not against it if the other options suck. It would be nice if the script could be fed through `-c` or via temp file that is recreated on every run, rather than potentially causing problems with stale cache.

martijnbastiaan, 2021-07-19:
> I'm surprised that a people are using the HEAD of cocotb. 

Well I'd usually advice against it, but we just got two PRs in so it's better than keeping our own fork I suppose.

> This seems like a convoluted fix, is there really no way to loop over the `VHD`L_LIB_ORDER` variable, compute the `VHDL_SOURCES_{lib}` variable name and get the value in Makefile's scripting language?

Absolutely, we could just loop over `VHDL_LIB_ORDER` in Make. However it wouldn't get you the "nice" error messages the Python implementation gives you - which would need some kind of set logic which I don't think is feasible within a Make. If this is what you prefer, I'll post a PoC tomorrow.

martijnbastiaan, 2021-07-20:
@ktbarrett I've made a solution based on Make in https://github.com/cocotb/cocotb/pull/2635, it even retains the error message (at a complexity cost though).

PR #2635 (2021-07-21): Add 'VHDL_LIB_ORDER' to specify VHDL library ordering for Xcelium/Questa/ModelSim
A regression test has been added to make sure we can now compile
multiple libraries. With four libraries the chance of a false positive
should be 1/4 * 1/3 * 1/2 ~ 4%.

Note that the logic sanity checking VHDL_LIB_ORDER and VHDL_SOURCES_*
has quadratic complexity, although it has been checked to perform within
bounds for at least 25 libraries.

Closes #2633
commit dc1c6417 (2021-07-21): Add 'VHDL_LIB_ORDER' to specify VHDL library ordering (#2635)

A regression test has been added to make sure we can now compile
multiple libraries. With four libraries the chance of a false positive
should be 1/4 * 1/3 * 1/2 ~ 4%.

Note that the logic sanity checking VHDL_LIB_ORDER and VHDL_SOURCES_*
has quadratic complexity, although it has been checked to perform within
bounds for at least 25 libraries.

Closes #2633
closed 2021-07-21 by ktbarrett