cocotb/cocotb #2636: Why so many test messages?

raysalemi, 2021-07-20:
Hi,

I'm writing a cocotb test that will loop through a list of test functions and call them. The test looks like this:

```python
@cocotb.test()
async def test_08_factory_tests(dut):
    """Tests different aspects of the factory"""
    tc08 = t08.s08_factory_classes_TestCase()
    methods = inspect.getmembers(t08.s08_factory_classes_TestCase)#, predicate=inspect.ismethod)
    for mm in methods:
        (name,function) = mm
        if name.startswith("test_"):
            test = getattr(tc08, name)
            tc08.setUp()
            print(f"**** Running: {name}")
            test()
            tc08.tearDown()
```

The functions run if they start with "test_"

```python
    def test_no_assert(self):
        pass

    def test_no_assert2(self):
        pass
```

Strangely, when I run this all the messages from cocotb get repeated once per function, even though they are all in one test:

```text
**** Running: test_no_assert
**** Running: test_no_assert2
  1000.00ns INFO     cocotb.regression                         regression.py:364  in _score_test                     Test Passed: test_08_factory_tests
INFO: regression.py(364)[cocotb.regression]: Test Passed: test_08_factory_tests
INFO: regression.py(364)[cocotb.regression]: Test Passed: test_08_factory_tests
  1000.00ns INFO     cocotb.regression                         regression.py:487  in _log_test_summary               Passed 1 tests (0 skipped)
INFO: regression.py(487)[cocotb.regression]: Passed 1 tests (0 skipped)
INFO: regression.py(487)[cocotb.regression]: Passed 1 tests (0 skipped)
```

And then later

```text
INFO: regression.py(557)[cocotb.regression]: **********************************************************************************************
** TEST                                  PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
**********************************************************************************************
** tinyalu_cocotb.test_08_factory_tests    PASS         1000.00          0.00   1533566.36  **
**********************************************************************************************

INFO: regression.py(557)[cocotb.regression]: **********************************************************************************************
** TEST                                  PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
**********************************************************************************************
** tinyalu_cocotb.test_08_factory_tests    PASS         1000.00          0.00   1533566.36  **
```

This is filling up my log with a lot of garbage.  Any idea why this is happening?

eric-wieser, 2021-07-20:
What is `t08.s08_factory_classes_TestCase`? Can you produce a minimal repro without it?

raysalemi, 2021-07-20:
Will do.

raysalemi, 2021-07-21:
I have simplified this issue and found something in the `pyuvm` code that I don't think should be causing this.

Here is my new test case:

```python
@cocotb.test()
async def test_08_factory(dut):
    """Tests different aspects of the factory"""
    print("****** Running test ******")
    tc08 = tests.TestHolder()
    methods = inspect.getmembers(tests.TestHolder)#, predicate=inspect.ismethod)
    for mm in methods:
        (name,function) = mm
        if name.startswith("test_"):
            test = getattr(tc08, name)
            tc08.setUp()
            print("******* Running ", name)
            test()
            tc08.tearDown()
    assert True
```

Here is the simplified `TestHolder`:

```python
from pyuvm import *
import pyuvm.utility_classes as utility_classes

class TestHolder:

    def setUp(self):
        self.top = uvm_component("top", None)

    def tearDown(self):
        uvm_root.clear_singletons()
        pass

    def test_no_assert(self):
        pass

    def test_no_assert2(self):
        pass
```

The above generates the spurious messages.  Why?  Because `uvm_component` extends `uvm_report_object` and `uvm_report_object` creates a logger and adds a `StreamHandler()` to it:

```python
    def __init__(self, name):
        super().__init__(name)
        full_name = self.get_full_name()
        self.logger = logging.getLogger(full_name)  # Every object gets its own logger
        self.logger.setLevel(level=logging.INFO)  # Default is to print INFO
        self.logger.propagate = False  # We are not sending log messages up the hierarchy
        handler = logging.StreamHandler()
        handler.setLevel(logging.NOTSET)  # Don't let the handler interfere with logger level
        # Make log messages look like UVM messages
        formatter = logging.Formatter("%(levelname)s: %(filename)s(%(lineno)d)[%(name)s]: %(message)s")
        handler.setFormatter(formatter)
        self.add_logging_handler(handler)
```

The bug goes away if I remove the last line here, so adding my `StreamHandler()` to this object's logger causes the messages.  I suspect I get one message each time I instantiate a `uvm_component()`

raysalemi, 2021-07-27:
Hi @eric-wieser, Any thoughts on why adding my `Streamhandler()` to my `logger` instance causes cocotb to print spurious messages from `regression.py`?  

Since I'm now basing **pyuvm** on **cocotb** perhaps I should use the **cocotb** logging system?  I'm not quite certain how I would do that. @ktbarrett, any thoughts on that?

raysalemi, 2021-07-27:
I've partially figured out the problem.  I'm getting so many messages because I was running the __init__ code above every time I ran a test, creating multiple uvm_components.  This is incorrect, I should only have been creating one.

It does not explain why **cocotb** sends messages to my logger.  Shouldn't **cocotb** have its own logging hierarchy?

raysalemi, 2021-07-27:
So the problem boils down to: every time I create my own `Logger` and my own `StreamHandler()` I get spurious messages from cocotb.  Any idea why and how to stop it?

eric-wieser, 2021-07-27:
What's the value of `full_name` When you call `logging.getLogger(full_name)`, you are asking to find an _existing_ logger with the name `full_name`, which might be one that cocotb is already using. It's your job as a user of the Python logging API to choose a suitable name.

raysalemi, 2021-07-27:
> What's the value of `full_name`? When you call `logging.getLogger(full_name)`, you are asking to find an _existing_ logger with the name `full_name`, which might be one that cocotb is already using. It's your job as a user of the Python logging API to choose a suitable name.

It is things such as this:

```text
FULL NAME: ''
FULL NAME: 'top'
FULL NAME: 'top.mid'
FULL NAME: 'top.sib'
FULL NAME: 'top.mid.orig'
FULL NAME: 'top.sib.orig'
FULL NAME: ''
FULL NAME: 'test'
FULL NAME: ''
FULL NAME: 'test_top'
FULL NAME: 'test_top.sib1'
FULL NAME: 'test_top.sib2'
```
They are all getting messages from cocotb.  

I wonder if the empty strings are doing it.

raysalemi, 2021-07-27:
Yup, @eric-wieser,  the zero length names were the problem.  This code ensures that there are no zero length logger names and that fixes the issue:

```python
        log_name = self.get_full_name()
        if len(log_name) == 0:
            log_name = "uvm_log_root"
        self.logger = logging.getLogger(log_name)
```

Thanks!
Ray

closed 2021-07-27 by raysalemi

eric-wieser, 2021-07-27:
Yes, the empty string is the global logger for which all other loggers are subloggers.

I recommend using `uvm` as a prefix for all of your loggers

eric-wieser, 2021-07-27:
That is, I would recommend doing:
```python
uvm_root_logger = logging.getLogger('uvm')
```
and then inside your `__init__`, replacing
```diff
-self.logger = logging.getLogger(full_name) 
+self.logger = uvm_root_logger .getChild('def.ghi')
```

raysalemi, 2021-07-27:
> That is, I would recommend doing:
>
> uvm_root_logger = logging.getLogger('uvm')
>
> and then inside your __init__, replacing
>
> -self.logger = logging.getLogger(full_name) +self.logger = uvm_root_logger.getChild(full_name)

Thanks. I’ll do that.

raysalemi, 2021-07-27:
> That is, I would recommend doing at the top of your file:
> 
> ```python
> uvm_root_logger = logging.getLogger('uvm')
> ```
> 
> and then inside your `__init__`, replacing
> 
> ```diff
> -self.logger = logging.getLogger(full_name) 
> +self.logger = uvm_root_logger.getChild(full_name)
> ```
> 
> If you attach the handlers to `uvm_root_logger`, then you also don't need to copy them to every logger.

Done.  Thanks again!