cocotb/cocotb #2643: Resulting Verilator C++ code does not compile

KGHVerilator, 2021-07-24:
I have a test that uses the xilinx xpm simple dual port ram. I did modify the xpm code to make it compile under verilator (remove the non-synthesizable stuff, fix up some of the changes in buffer lengths, specify explicit types for parameters). 
This code compiles and builds a working simulation with verilator on its own. However when I try to build with cocotb I get compile errors at the C++ compile level.

cocotb:simulator.py:256 Running command: perl /usr/bin/verilator -cc --exe -Mdir /home/hamilton/workspace/vtr_ev_top/logic/common_sv/sed_ram/cocotb/sim_build/simple_dual_port_ram_tb/RAM_WIDTH=8_DEPTH_ORDER=4_READ_LATENCY=1 -DCOCOTB_SIM=1 --top-module simple_dual_port_ram_tb --vpi --public-flat-rw --prefix Vtop -o simple_dual_port_ram_tb -LDFLAGS -Wl,-rpath,/usr/local/lib/python3.8/dist-packages/cocotb/libs -L/usr/local/lib/python3.8/dist-packages/cocotb/libs -lcocotbvpi_verilator -I/home/hamilton/workspace/vtr_ev_top/logic --relative-includes +1800-2017ext+sv +1364-2005ext+v --trace-fst --trace-structs -GRAM_WIDTH=8 -GDEPTH_ORDER=4 -GREAD_LATENCY=1 /usr/local/lib/python3.8/dist-packages/cocotb/share/lib/verilator/verilator.cpp /home/hamilton/workspace/vtr_ev_top/logic/common_sv/sed_ram/cocotb/simple_dual_port_ram_tb.sv /home/hamilton/workspace/vtr_ev_top/logic/verilator_common/src/unisims/FDRE.v /home/hamilton/workspace/vtr_ev_top/logic/verilator_common/src/unisims/FDSE.v /home/hamilton/workspace/vtr_ev_top/logic/verilator_common/src/unisims/USR_ACCESSE2.v /home/hamilton/workspace/vtr_ev_top/logic/verilator_common/src/unisims/DSP48E1.sv /home/hamilton/workspace/vtr_ev_top/logic/verilator_common/src/unisims/DNA_PORT.v /home/hamilton/workspace/vtr_ev_top/logic/verilator_common/src/xpm/xpm_cdc/hdl/xpm_cdc.sv /home/hamilton/workspace/vtr_ev_top/logic/verilator_common/src/xpm/xpm_fifo/hdl/xpm_fifo.sv /home/hamilton/workspace/vtr_ev_top/logic/verilator_common/src/xpm/xpm_memory/hdl/xpm_memory.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/sed_ram/single_port_ram.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/sed_ram/simple_dual_port_ram.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/flipflop/ff_std_logic.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/flipflop/ff_vec.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/clock_cross/double_flop_vec.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/clock_cross/double_flop.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/sed_fifo/sync_fifo.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/axi_lite_mnc_write_clk_change.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/axi_mnc.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/mnc_regmap_load_pipe.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/axi_stream_slave_to_file.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/axilite_regmap_load_file_handler.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/axilite_regmap_load_pipe.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/dynamic_iq_read.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/dynamic_iq_write.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/dynamic_vector_read.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/dynamic_vector_write.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/get_sim_config_value.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/read_finish_from_pipe.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/read_vector_from_pipe.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/software_capture_vector_to_pipe.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/send_ack_to_pipe.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/triggered_read_vector_from_pipe.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/mnc/write_iq_to_pipe.sv /home/hamilton/workspace/vtr_ev_top/logic/common_sv/monitor/unique_id.sv
INFO     cocotb:simulator.py:256 Running command: make -C /home/hamilton/workspace/vtr_ev_top/logic/common_sv/sed_ram/cocotb/sim_build/simple_dual_port_ram_tb/RAM_WIDTH=8_DEPTH_ORDER=4_READ_LATENCY=1 -f Vtop.mk
INFO     cocotb:simulator.py:267 make: Entering directory '/home/hamilton/workspace/vtr_ev_top/logic/common_sv/sed_ram/cocotb/sim_build/simple_dual_port_ram_tb/RAM_WIDTH=8_DEPTH_ORDER=4_READ_LATENCY=1'
INFO     cocotb:simulator.py:267 ccache g++  -std=c++11 -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o Vtop__ALL.o Vtop__ALL.cpp
INFO     cocotb:simulator.py:267 In file included from Vtop__ALL.cpp:18:
INFO     cocotb:simulator.py:267 Vtop__Syms.cpp: In constructor ‘Vtop__Syms::Vtop__Syms(VerilatedContext*, const char*, Vtop*)’:
INFO     cocotb:simulator.py:267 Vtop__Syms.cpp:308:157: error: ‘class Vtop___024root’ has no member named ‘var_simple_dual_port_ram_tb__DOT__uut__DOT__inst__DOT__xpm_memory_base_inst__DOT__P_ECC_MODE’; did you mean ‘var_simple_dual_port_ram_tb__DOT__uut__DOT__inst__DOT__xpm_memory_base_inst__DOT__ECC_MODE’?
INFO     cocotb:simulator.py:267   308 |         __Vscope_simple_dual_port_ram_tb__uut__inst__xpm_memory_base_inst.varInsert(__Vfinal,"P_ECC_MODE", const_cast<void*>(static_cast<const void*>(&(TOP.var_simple_dual_port_ram_tb__DOT__uut__DOT__inst__DOT__xpm_memory_base_inst__DOT__P_ECC_MODE))), true, VLVT_WDATA,VLVD_NODIR|VLVF_PUB_RW,1 ,175,0);
INFO     cocotb:simulator.py:267       |                                                                                                                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
INFO     cocotb:simulator.py:267       |                                                                                                                                                             var_simple_dual_port_ram_tb__DOT__uut__DOT__inst__DOT__xpm_memory_base_inst__DOT__ECC_MODE
INFO     cocotb:simulator.py:267 Vtop__Syms.cpp:313:165: error: ‘class Vtop___024root’ has no member named ‘var_simple_dual_port_ram_tb__DOT__uut__DOT__inst__DOT__xpm_memory_base_inst__DOT__P_MEMORY_PRIMITIVE’; did you mean ‘var_simple_dual_port_ram_tb__DOT__uut__DOT__inst__DOT__xpm_memory_base_inst__DOT__MEMORY_PRIMITIVE’?
INFO     cocotb:simulator.py:267   313 |         __Vscope_simple_dual_port_ram_tb__uut__inst__xpm_memory_base_inst.varInsert(__Vfinal,"P_MEMORY_PRIMITIVE", const_cast<void*>(static_cast<const void*>(&(TOP.var_simple_dual_port_ram_tb__DOT__uut__DOT__inst__DOT__xpm_memory_base_inst__DOT__P_MEMORY_PRIMITIVE))), true, VLVT_WDATA,VLVD_NODIR|VLVF_PUB_RW,1 ,87,0);
INFO     cocotb:simulator.py:267       |                                                                                                                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
INFO     cocotb:simulator.py:267       |                                                                                                                                                                     var_simple_dual_port_ram_tb__DOT__uut__DOT__inst__DOT__xpm_memory_base_inst__DOT__MEMORY_PRIMITIVE
INFO     cocotb:simulator.py:267 make: *** [/usr/share/verilator/include/verilated.mk:255: Vtop__ALL.o] Error 1
INFO     cocotb:simulator.py:267 make: Leaving directory '/home/hamilton/workspace/vtr_ev_top/logic/common_sv/sed_ram/cocotb/sim_build/simple_dual_port_ram_tb/RAM_WIDTH=8_DEPTH_ORDER=4_READ_LATENCY=1'
ERROR    cocotb:simulator.py:271 Command terminated with error 2

KGHVerilator, 2021-07-24:
Here is a log of building with verilator/c++ only (this works)
ninja -v
[1/11] ccache /usr/bin/cc  -I/home/hamilton/workspace/d4_decimator_top/logic/src/fft/verilator -I/home/hamilton/workspace/d4_decimator_top/logic -I/home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/include -isystem /usr/share/verilator/include/vltstd -isystem /usr/local/share/verilator/include/vltstd -g -MD -MT /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/dynamic_iq_read.c.o -MF /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/dynamic_iq_read.c.o.d -o /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/dynamic_iq_read.c.o   -c /home/hamilton/workspace/d4_decimator_top/logic/common_sv/mnc/dpi/dynamic_iq_read.c
[2/11] ccache /usr/bin/cc  -I/home/hamilton/workspace/d4_decimator_top/logic/src/fft/verilator -I/home/hamilton/workspace/d4_decimator_top/logic -I/home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/include -isystem /usr/share/verilator/include/vltstd -isystem /usr/local/share/verilator/include/vltstd -g -MD -MT /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/get_sim_config_value.c.o -MF /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/get_sim_config_value.c.o.d -o /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/get_sim_config_value.c.o   -c /home/hamilton/workspace/d4_decimator_top/logic/common_sv/mnc/dpi/get_sim_config_value.c
[3/11] ccache /usr/bin/cc  -I/home/hamilton/workspace/d4_decimator_top/logic/src/fft/verilator -I/home/hamilton/workspace/d4_decimator_top/logic -I/home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/include -isystem /usr/share/verilator/include/vltstd -isystem /usr/local/share/verilator/include/vltstd -g -MD -MT /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/axilite_regmap_load_file_handler.c.o -MF /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/axilite_regmap_load_file_handler.c.o.d -o /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/axilite_regmap_load_file_handler.c.o   -c /home/hamilton/workspace/d4_decimator_top/logic/common_sv/mnc/dpi/axilite_regmap_load_file_handler.c
[4/11] ccache /usr/bin/cc  -I/home/hamilton/workspace/d4_decimator_top/logic/src/fft/verilator -I/home/hamilton/workspace/d4_decimator_top/logic -I/home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/include -isystem /usr/share/verilator/include/vltstd -isystem /usr/local/share/verilator/include/vltstd -g -MD -MT /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/read_vector_from_pipe.c.o -MF /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/read_vector_from_pipe.c.o.d -o /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/read_vector_from_pipe.c.o   -c /home/hamilton/workspace/d4_decimator_top/logic/common_sv/mnc/dpi/read_vector_from_pipe.c
[5/11] ccache /usr/bin/cc  -I/home/hamilton/workspace/d4_decimator_top/logic/src/fft/verilator -I/home/hamilton/workspace/d4_decimator_top/logic -I/home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/include -isystem /usr/share/verilator/include/vltstd -isystem /usr/local/share/verilator/include/vltstd -g -MD -MT /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/send_ack_to_pipe.c.o -MF /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/send_ack_to_pipe.c.o.d -o /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/send_ack_to_pipe.c.o   -c /home/hamilton/workspace/d4_decimator_top/logic/common_sv/mnc/dpi/send_ack_to_pipe.c
[6/11] ccache /usr/bin/cc  -I/home/hamilton/workspace/d4_decimator_top/logic/src/fft/verilator -I/home/hamilton/workspace/d4_decimator_top/logic -I/home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/include -isystem /usr/share/verilator/include/vltstd -isystem /usr/local/share/verilator/include/vltstd -g -MD -MT /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/write_iq_to_pipe.c.o -MF /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/write_iq_to_pipe.c.o.d -o /home/hamilton/workspace/d4_decimator_top/logic/verilator_common/build/common_sv/mnc_dpi/CMakeFiles/common_sv_mnc_dpi.dir/write_iq_to_pipe.c.o   -c /home/hamilton/workspace/d4_decimator_top/logic/common_sv/mnc/dpi/write_iq_to_pipe.c
[7/11] cd /home/hamilton/workspace/d4_decimator_top/logic/src/fft/verilator && /usr/bin/verilator_bin --compiler gcc --prefix Vaccum_impl_tb --Mdir /home/hamilton/workspace/d4_decimator_top/logic/src/fft/verilator/build/CMakeFiles/a [... cut ...]   /home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/lib/libeah.a  /usr/lib/x86_64-linux-gnu/libboost_iostreams.so.1.71.0  /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0  -lpthread  /usr/lib/x86_64-linux-gnu/libboost_atomic.so.1.71.0  /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.71.0  /usr/lib/x86_64-linux-gnu/libz.so  /usr/lib/x86_64-linux-gnu/libgpiodcxx.so  /usr/lib/x86_64-linux-gnu/libftdi1.so  /usr/lib/x86_64-linux-gnu/libusb-1.0.so  /usr/lib/x86_64-linux-gnu/libxml2.so && :

KGHVerilator, 2021-07-24:
Here is the python file that is building my simple cocotb test (it instantiates the memory to show the failure of the generated C++ code).

#!/usr/bin/env python3

import os
from typing import List
import pytest
from cocotb_test.simulator import Verilator

def logicTopDir() -> str:
    path = os.path.dirname(os.path.abspath(__file__))
    return os.path.realpath(path + "/../../../")

def verilatorCommonFiles(logicTop: str) -> List[str]:
    verilatorCommonDir = logicTop + "/verilator_common"
    retval = []
    retval.append(verilatorCommonDir +"/src/unisims/FDRE.v")
    retval.append(verilatorCommonDir +"/src/unisims/FDSE.v")
    retval.append(verilatorCommonDir +"/src/unisims/USR_ACCESSE2.v")
    retval.append(verilatorCommonDir +"/src/unisims/DSP48E1.sv")
    retval.append(verilatorCommonDir +"/src/unisims/DNA_PORT.v")
    retval.append(verilatorCommonDir +"/src/xpm/xpm_cdc/hdl/xpm_cdc.sv")
    retval.append(verilatorCommonDir +"/src/xpm/xpm_fifo/hdl/xpm_fifo.sv")
    retval.append(verilatorCommonDir +"/src/xpm/xpm_memory/hdl/xpm_memory.sv")

    return retval

def commonSvFiles(logicTop: str) -> List[str]:
    commonSvDir = logicTop + "/common_sv"
    retval = []
    retval.append(commonSvDir + "/sed_ram/single_port_ram.sv")
    retval.append(commonSvDir + "/sed_ram/simple_dual_port_ram.sv")
    retval.append(commonSvDir + "/flipflop/ff_std_logic.sv")
    retval.append(commonSvDir + "/flipflop/ff_vec.sv")
    retval.append(commonSvDir + "/clock_cross/double_flop_vec.sv")
    retval.append(commonSvDir + "/clock_cross/double_flop.sv")
    retval.append(commonSvDir + "/sed_fifo/sync_fifo.sv")

    retval.append(commonSvDir + "/mnc/axi_lite_mnc_write_clk_change.sv")
    retval.append(commonSvDir + "/mnc/axi_mnc.sv")
    retval.append(commonSvDir + "/mnc/mnc_regmap_load_pipe.sv")
    retval.append(commonSvDir + "/mnc/axi_stream_slave_to_file.sv")
    retval.append(commonSvDir + "/mnc/axilite_regmap_load_file_handler.sv")
    retval.append(commonSvDir + "/mnc/axilite_regmap_load_pipe.sv")
    retval.append(commonSvDir + "/mnc/dynamic_iq_read.sv")
    retval.append(commonSvDir + "/mnc/dynamic_iq_write.sv")
    retval.append(commonSvDir + "/mnc/dynamic_vector_read.sv")
    retval.append(commonSvDir + "/mnc/dynamic_vector_write.sv")
    retval.append(commonSvDir + "/mnc/get_sim_config_value.sv")
    retval.append(commonSvDir + "/mnc/read_finish_from_pipe.sv")
    retval.append(commonSvDir + "/mnc/read_vector_from_pipe.sv")
    retval.append(commonSvDir + "/mnc/software_capture_vector_to_pipe.sv")
    retval.append(commonSvDir + "/mnc/send_ack_to_pipe.sv")
    retval.append(commonSvDir + "/mnc/triggered_read_vector_from_pipe.sv")
    retval.append(commonSvDir + "/mnc/write_iq_to_pipe.sv")
    retval.append(commonSvDir + "/monitor/unique_id.sv")
    return retval

def verilatorCompileArgs(logicTop: str) -> List[str]:
    retval = []
    retval.append("-I" + logicTop)
    retval.append("--relative-includes")
    retval.append("+1800-2017ext+sv")
    retval.append("+1364-2005ext+v")

    return retval

@pytest.mark.parametrize("testParameters", [{"RAM_WIDTH": "8", "DEPTH_ORDER" : 4, "READ_LATENCY" : 1}])
def testLnbSpiInterface(testParameters):
    path = os.path.dirname(os.path.abspath(__file__))

    sources = [path + "/simple_dual_port_ram_tb.sv"] + verilatorCommonFiles(logicTopDir()) + commonSvFiles(logicTopDir())
    sim = Verilator(
        toplevel="simple_dual_port_ram_tb",
        module="simple_dual_port_ram_tb",
        compile_args=verilatorCompileArgs(logicTopDir()),
        verilog_sources=sources,
        parameters=testParameters,
        sim_build="sim_build/simple_dual_port_ram_tb/" + "_".join(("{}={}".format(*i) for i in testParameters.items())),
        waves=1
    )

    sim.run()

KGHVerilator, 2021-07-24:
I noticed in the c++ command line by verilator you have -std=c++11 and -std=gnu++14

imphil, 2021-07-25:
Which Verilator version are you using? Only Verilator 4.106 is supported by cocotb currently. In more recent versions, the access to internal signals changed (and before that, changes to VPI broke the integration). (See also https://github.com/cocotb/cocotb/issues/2300)

KGHVerilator, 2021-07-26:
I was using verilator 4.210, Sorry I do not see where it says 4.106. It does work with verilator 4.106.

closed 2021-07-26 by KGHVerilator