cocotb/cocotb #2646: Make Example that uses IP

ktbarrett, 2021-07-25:
Several users have asked for an example of how to use the makefiles in projects that contain precompiled and/or encrypted IP cores or libraries. There should be a simple example. It will likely not work for them, but could act as a template.

Alternatively, documentation or an explicit API for these objects would also help.

kuriousd, 2025-01-29:
Hi!

I've got some stuff on this:

- Xilinx FFT core working with modelsim/questa
- UNISIM components working with GHDL

Could this help? I've seen in cocotb 2.0 presentation makefiles will be deprecated so maybe I could add the example after 2.0 merge.

ktbarrett, 2025-01-29:
Both would absolutely help! If they are standalone projects, rather than heap them into the cocotb repo, I think we would want to fork a copy into our org and have the cocotb documentation point to them.

cmarqu, 2025-01-29:
> I've seen in cocotb 2.0 presentation makefiles will be deprecated so maybe I could add the example after 2.0 merge.

The replacement for Makefiles, the Python Runner, is already present in 1.9, so you could already go ahead now: https://docs.cocotb.org/en/stable/runner.html