cocotb/cocotb #2648: When re-running xcelium simulation cocotb is not running.

ronple, 2021-08-01:
Hi,

After running xcelium simulation I re-run simulation but cocotb testbench is not driving any signal as it should.
Here is the run info:    
- cocotb:  _1.5.2_
- OS: _centos7_
- xcelium:  _xrun(64)  20.09-s002:_
- python: _3.6_
- log file:  [sim.log](https://github.com/cocotb/cocotb/files/6912429/sim.log)

Is the fact that the log shows that it is shutting down after the 1st run causing the re-run not to work after it?
is there a way to change it?

Regards,
ronple

cmarqu, 2021-08-02:
The "Shutting down" is a harmless exit message.
Do the tests and examples of cocotb run successfully for you?

ronple, 2021-08-02:
My test is running successfully.
The problem is that once the test is finished I am re-running it from the xcelium simulation side and the TB is not running at all.

ktbarrett, 2021-08-02:
See the [startup hook](https://github.com/cocotb/cocotb/blob/2b5b820c4ded3cb22852993471dcb99b1eadf326/cocotb/share/lib/vpi/VpiCbHdl.cpp#L469-L476). Looks like Xcelium/IUS does something different for startup. Not sure why, there is no note. You could try removing that condition and see if that works? Or there is `vpiStartOfRestart` that might also work.

marlonjames, 2021-08-02:
It's difficult for VPI/VHPI applications when the simulators don't support the necessary callbacks...

ktbarrett, 2024-08-17:
Rerunning simulations is not a feature that cocotb currently supports.

closed 2024-08-17 by ktbarrett