cocotb/cocotb #2649: Fork not operating as expected.

raysalemi, 2021-08-01:
I don't know why I'm seeing fork refuse to return until its coroutine  Is finished.  I have this coroutine:

```pyton

async def waitabit(abit=5):
    Timer(1, units="us")

    @staticmethod
    async def do_nonblocking_put(put_port, data_list):
        for data in data_list:
            ii = 0
            while not put_port.try_put(data) and ii < 5:
                print(data, put_port)
                await waitabit()
                ii += 1

# later I fork off the routine to put the data in the FIFO. This should loop until I launch the getter.

      put_data = [10, 20, 30, 'c0', None]
      pp = uvm_nonblocking_put_port("pp", self.my_root)
      cocotb.fork(self.do_nonblocking_put(pp, put_data))
      print("MADE IT")
```

However, the fork containing `do_nonblocking_put` does not return until it has completed its loop. I had to place `ii` in there to avoid an infinite loop:

Notice that the `10` gets placed in the FIFO and then the coroutine gets stuck trying to put the rest of them. It is supposed to get stuck, but it `fork` is also supposed to return so I can launch the consumer.

```text
20 my_root.pp
20 my_root.pp
20 my_root.pp
20 my_root.pp
20 my_root.pp
30 my_root.pp
30 my_root.pp
30 my_root.pp
30 my_root.pp
30 my_root.pp
c0 my_root.pp
c0 my_root.pp
c0 my_root.pp
c0 my_root.pp
c0 my_root.pp
None my_root.pp
None my_root.pp
None my_root.pp
None my_root.pp
None my_root.pp
MADE IT
```

`MADE IT` should have been printed immediately upon `fork`'s immediate return. But `fork` is not returning until the coroutine returns. I cannot see that I'm doing anything wrong.

eric-wieser, 2021-08-01:
Did you mean to write
```python
async def waitabit(abit=5):
    await Timer(1, units="us")
```
instead of what you actually wrote,
```python
async def waitabit(abit=5):
    Timer(1, units="us")
```

eric-wieser, 2021-08-01:
@ktbarrett, this is why I think it is a bad idea to conflate `Time` and `Timer` types - we could make cocotb emit a warning if a trigger is destroyed but never awaited, but we can't do that if the triggers are also just datatypes.

raysalemi, 2021-08-01:
> Did you mean to write
> 
> ```python
> async def waitabit(abit=5):
>     await Timer(1, units="us")
> ```
> 
> instead of what you actually wrote,
> 
> ```python
> async def waitabit(abit=5):
>     Timer(1, units="us")
> ```
> 
> fork is returning immediately because the coroutine you're running is finishing immediately, as you never told it to wait.

DOH!!! 

I agree with @eric-wieser that there are too many people like me who would make this mistake.

raysalemi, 2021-08-01:
It works now.

raysalemi, 2021-08-01:
Thanks for the fast reply!

closed 2021-08-01 by raysalemi