cocotb/cocotb #2651: cocotb importing gives error on simutor importing

haykp, 2021-08-04:
I saw several issues related to this, however all these issues were asking to update the cocotb. 
We tried with the latest cocotb package, after installation when we import it gives this error:

Python 3.9.6 (default, Aug  4 2021, 16:38:11)
[GCC 4.8.5 20150623 (Red Hat 4.8.5-36)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cocotb
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/fdesign/u0101652/cocotb-master/cocotb/__init__.py", line 43, in <module>
    import cocotb.handle
  File "/home/fdesign/u0101652/cocotb-master/cocotb/handle.py", line 39, in <module>
    from cocotb import simulator
ImportError: cannot import name 'simulator' from partially initialized module 'cocotb' (most likely due to a circular import) (/home/fdesign/u0101652/cocotb-master/cocotb/__init__.py)
>>

Can someone please help understand how to solve this issue?

The cocotb installation was done using setup.py program.
./setup.py build
./setup.py install 
( in our company due to security we cannot use pip, as it needs internet access)

eric-wieser, 2021-08-04:
> in our company due to security we cannot use pip, as it needs internet access

Are you permitted to use `pip install . --no-index`, which does not use internet access? That may or may not actually help, of course.

eric-wieser, 2021-08-04:
Yes, that is what I meant. I guess `--no-index` is more complex than I thought. Passing `--no-build-isolation` as well may or may not help.

What version of `setuptools` do you have installed?

eric-wieser, 2021-08-04:
Wait a minute:
```
 Complete output from command /cae/swdev/sdk/python/3.7.2/Linux7.6/bin/python3.7 -m pip
 ```
 vs
 ```
Python 3.9.6 (default, Aug  4 2021, 16:38:11)
[GCC 4.8.5 20150623 (Red Hat 4.8.5-36)] on linux
```
It looks like you installed cocotb into one python but are somehow trying to import it from a different one?

haykp, 2021-08-04:
Good catch, it is actually my fault when asking. 
I tried with both 3.9.6 and 3.7 version. So for both of them not working

eric-wieser, 2021-08-04:
You should not be in the `cocotb-master` directory when trying to import cocotb

haykp, 2021-08-05:
this issue is resolved.
After going out from cocotb-master it worked.

So most important for offline cocotb installation not to forget to download/install all the required packages.

Now cocotb importing works, but other packages like cocotb-axi does not work.

closed 2021-08-05 by haykp