cocotb/cocotb #2654: VPI: Implementation does not know how to iterate over vpiInterface(601)

avidan-efody, 2021-08-09:
vpiInterface is not mentioned here: https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/vpi/VpiCbHdl.cpp#L546

As a result, we can't iterate over interface children.

Is there any objection to adding interface childrens in the location above? the relevant VPI object hierarchy diagrams are in section 37.10 37.6 of ieee 1800-2012

ktbarrett, 2021-08-09:
No we have nothing against adding SV VPI extensiom support. Feel free to open a PR. Heads up though, it may not be a simple task. I have no idea how simulators that don't support the SV VPI extensions will handle it.

avidan-efody, 2021-08-10:
We could also use the vpiInstance which is part of verilog standard