cocotb/cocotb #2655: cocotb.queue.queue and clocked blocking

raysalemi, 2021-08-09:
The code below suggests to me that **cocotb** would benefit from a `cput` and `cget` function which means clocked put and clocked get.  The idea would be to take the loop below and provide it as a method.

It might look like this:

```
qq = cocotb.queue.Queue()
qq.cput(dut.clk, item)
```

I'm going to try this out in UVMQueue. Blocking is an essential part of the UVM's use model. 

```python
    async def send_op(self, aa, bb, op):
        self.driver_queue.put((aa, bb, op))
```

You can see the blocking put.  I had no idea that the block would stop the whole simulator, I figured it would only stop this thread, a SystemVerilog paradigm.

The new one looks like this:

```python
    async def send_op(self, aa, bb, op):
        while True:
            await FallingEdge(self.dut.clk)
            try:
                self.driver_queue.put_nowait((aa, bb, op))
                return
            except QueueFull:
                continue
```

Thank you all!

_Originally posted by @raysalemi in https://github.com/cocotb/cocotb/issues/2653#issuecomment-895096434_

eric-wieser, 2021-08-09:
Did you mean to put the `await` before `self.driver_queue.put` there?

raysalemi, 2021-08-09:
> Did you mean to put an `await` before `self.driver_queue.put` there?

Yes. My systemVerilog is showing.

Python is much better than SystemVerilog in that way as well in that SystemVerilog has no syntactic way of telling whether you are calling a function or task.

raysalemi, 2021-08-09:
@eric-wieser , I realized that the `cput` name is unneeded.  The UVMQueue will simply require a trigger for put and get and will only "block" using the clock.

ktbarrett, 2021-08-09:
@raysalemi you should see a bunch of warnings in your log coming from Python about coroutines that were never awaited. That can help you diagnose these problems.

raysalemi, 2021-08-09:
I've now created UVMQueue and UVMEvent that loop on clocking events to do put, get, and wait.

```python
class UVMEvent(Event):

    def __init__(self, name, clock):
        super().__init__(name=name)
        self.trigger = Timer(10, units="step")
    
    async def wait(self):
        while True:
            await self.trigger
            if self.is_set:
                return self.fired
            else:
                continue
```

raysalemi, 2021-08-09:
I had fixed all of those.

closed 2021-08-09 by raysalemi