cocotb/cocotb #2656: Clock event put, get, and wait

raysalemi, 2021-08-09:
I found that my simulation was hanging when I called queue.put on a queue with maxlen=1.  I also hung awaiting Event.wait()

I've fixed the problem by creating versions of the queue and event that loop on a clock-based event so that control goes back to the simulator.   Now you pass `uvm_root().run_test()` the clock as well as the test name. This gets stored in ConfigDB() and can be used anywhere in the testbench.

## Clock-based blocking

Here is the event:

```python
class UVMEvent(Event):

    def __init__(self, name, clock):
        super().__init__(name=name)
        self.trigger = Timer(10, units="step")
    
    async def wait(self):
        while True:
            await self.trigger
            if self.is_set:
                return self.fired
            else:
                continue
```

And here are the `put` and `get`:

```python
    async def put(self, item):
        while True:
            await self.trigger
            try:
                self.put_nowait(item)
                break
            except QueueFull:
                continue        

    async def get(self):
        while True:
            await self.trigger
            try:
                item =  self.get_nowait()
                print("GOT:", item)
                return item
            except QueueEmpty:
                continue
```

Do you think this a generalizable feature in cocotb?

marlonjames, 2021-08-09:
There is no inherit problem with a task calling `await q.put()` on a `Queue` with `maxlen = 1`. The calling task will suspend until there is room in `q`, which can only happen if another task calls `await q.get()` or `q.get_nowait()`.

I don't see value in adding extra `Timer` triggers or waiting on clock edges inside these coroutines, as they are already waiting on some `Event` to be triggered.
If your simulation is not progressing, there is some other underlying issue.

raysalemi, 2021-08-09:
It’s my experience that await q.put will hang the simulator as there is no
clocking event. The clock stops and the queues cannot be emptied.

raysalemi, 2021-08-09:
> There is no inherit problem with a task calling `await q.put()` on a `Queue` with `maxlen = 1`. The calling task will suspend until there is room in `q`, which can only happen if another task calls [...]

I thought that one could not block on qq.put(), for example and keep the simulator running. I believe one has to await a clocked event to give control back to the simulator.

This solution allows the `put()` and `get()` functionality that one expects from a Queue without blocking the simulator.  

If I'm missing something I'd love to know since I plan to teach others how to do this and I don't want to get it wrong.

ktbarrett, 2021-08-09:
> I believe one has to await a clocked event to give control back to the simulator.

Why do you need to give control back to the simulator? On the other end of the queue should be a different forked task that is, or will soon be, pulling messages from that queue.

raysalemi, 2021-08-09:
It does not because the simulator clock is stopped and the forked task
loops on FallingEdge.

eric-wieser, 2021-08-09:
I think there's a misunderstanding here. Control is returned to the simulator simply when there are no coroutines left to run, right? The scheduler loop is roughly:

* Simulator triggers a "real" event and wakes up some coroutines
* Those coroutines set events and may wake up other coroutines
* Eventually the last coroutine `await`s and control returns to the simulator

The only thing that makes "real" triggers special is that they _start_ the chain of events. Any type of trigger can end a chain of events.

If you end up in a situation where the only events waiting to fire are "virtual", your testbench has a deadlock and is itself the problem; no change to your dut can ever wake it.

ktbarrett, 2021-08-09:
> your testbench has an infinite loop in which no time passes

I'm starting to think this is the real problem. IIRC I remember seeing some polling loops.

raysalemi, 2021-08-09:
> > your testbench has an infinite loop in which no time passes
> 
> 
> 
> I'm starting to think this is the real problem. IIRC I remember seeing some polling loops.

Those are gone, if you mean the ones in the queue.

eric-wieser, 2021-08-09:
If the symptom is that `cocotb.Clock` seems to freeze, then @ktbarrett's diagnosis is correct. Either:

* You have an infinite loop in your testbench
* There is a real bug in the scheduler, and we need a more minimal example to track it down

raysalemi, 2021-08-09:
@eric-wieser, your description is how I *thought* it worked, but it doesn't work that way.

My hanging version of the testbench had a forked coroutine that looped on `FallingEdge`  It read data from a queue using `get.no_wait()` and if there was no data it looped around to the next falling edge.

I had another forked coroutine that put data into the queue.  I figured I'd put one item into the one-deep queue, loop around, and block on the next put.  Then the simulator would let the consumer run.

This is not what happened .  Instead the blocking put caused the simulator to stop running.  This meant there was no falling edge and the queue was never emptied.   **deadlock**.

@marlonjames said that this was because you needed to await a clocked coroutine to get the simulator running again. I implemented the above based on this understanding and it works.

If the thought is that a blocking put on a cocotb.queue.Queue should *not* block the simulator then there is something wrong because there was no way to make that work.

raysalemi, 2021-08-09:
Perhaps the problem is that the `uvm_root().run_test()` coroutine used  a coroutine Event and blocked until all the objections to ending had been dropped.  Can `Event.wait()` block the simulator?

eric-wieser, 2021-08-09:
Are you able to produce a minimal reproduction using as few components of PyUVM as possible?

eric-wieser, 2021-08-09:
> Can Event.wait() block the simulator?

Something like the following will block the simulator:
```
async def test():
    e1 = Event()
    e2 = Event()
    def t1():
        while True:
            e1.set()
            await e2
            e2.clear()
    def t2():
        while True:
            await e1
            e1.clear()
            e2.set()
    fork(t1())
    fork(t2())
    await Timer(1)
    # never reached
```
But this is by design.

raysalemi, 2021-08-09:
I have figured out that the block was actually being cased by the Event in a situation similar to what you've written above.  The main thread runs and launches all the run_phase threads in their own fork. Then it blocks until all the run phases are finished. (all objections dropped.)  There was a bug in that code that was causing the event not be set and cleared properly, and that's where I think the hang was.

I've gone back to an earlier version, fixed only that portion and that fixed the hang.

So I'll be backing out all these changes to the queue.

Thank you!

Ray

closed 2021-08-15 by raysalemi