cocotb/cocotb #2658: Operator precedence of `<=` misleading

martijnbastiaan, 2021-08-10:
In Python `<=` has higher precedence than `not`, `or`, `and`, `if-then`, and `lambda`. In cocotb `<=` is reused as an assignment operator, which leads to some highly surprising behavior when compared to the "normal" assignment operator `=`.

|   | Example              | Expected               | Actual                 |
|---|----------------------|------------------------|------------------------|
| 1 | `a <= b or c`        | `a <= (b or c)`        | `(a <= b) or c`        |
| 2 | `a <= b and c`       | `a <= (b and c)`       | `(a <= b) and c`       |
| 3 | `a <= b if c else d` | `a <= (b if c else d)` | `(a <= b) if c else d` |

I recently had my third encounter with this behavior. Although I had seen it twice before, it still took me some time to realize it was an operator precedence issue thanks to the pervasiveness of "normal" assignments.

# Partial solution
To prevent (1) and (2) `<=` could return a singleton `DontUse` like:

```python
class DontUse:
    def __bool__(self):
        raise ValueError("Don't use me like this dummy")

DontUse = DontUse()
```

This would make Python throw the error when trying to evaluate `or` and `and`. There's no such workaround available for `if-then` AFAIK.

# Full solution
I don't think there's much we can do. I think using `<=` should be deprecated and eventually removed entirely. There's very little benefit, but it can cause some pretty annoying debug sessions.  Assignment on simulator objects should probably replace `<=`'s current function. 

| Old               | New                |
|-------------------|--------------------|
| `dut.a <= b or c` | `dut.a = b or c`   |
| `a <= c`          | `a.value = b or c` |

The good news is that there's a good upgrade path:

1. Deprecation warning on `<=`
2. Hard error in `__le__`
3. Removal of `__le__` (?)

# Workarounds
1. Use parentheses:  `a <= (b or c)`
2. Use `.value`: `a.value = b or c`

eric-wieser, 2021-08-10:
We already implement the partial solution you suggest:

https://github.com/cocotb/cocotb/blob/fba7b854b00375500552ff8804757f468152b111/cocotb/handle.py#L476-L483

https://github.com/cocotb/cocotb/blob/fba7b854b00375500552ff8804757f468152b111/cocotb/handle.py#L418-L434

eric-wieser, 2021-08-10:
Were you on an old version of cocotb? Was the error being silenced somehow?

martijnbastiaan, 2021-08-10:
My last two (recent) run-ins were with `if-then`. I've also seen it with `or`/`and` but that has been a while ago, so that very well might have been an old version of cocotb. Good to know that that safety net is there, and I should have checked before creating this issue of course :-).

ktbarrett, 2021-08-10:
I would definitely prefer the `.value` syntax BTW. `<=` is "neat" but obviously has issues. We recently deprecated setattr syntax and I think we should also deprecate `<=` syntax. Thoughts?

> There should be one-- and preferably only one --obvious way to do it.

martijnbastiaan, 2021-08-10:
Works for me. It's non-surprising, performant, and solves this issue. It's slightly annoying that it makes testbenches more verbose, but I :100:% prefer that over the status quo.

closed 2021-08-21 by ktbarrett

o-alexandre-felipe, 2024-02-26:
I will miss the clean verilog like `<=`

PR #2667 (2021-08-21): Deprecate `<=` syntax for setting values on handles
Closes #2658. Closes #2681.