cocotb/cocotb #2659: `gather` and `select` to replace `Combine` and `First`

ktbarrett, 2021-08-11:
`Combine` and `First` only work on triggers and tasks, not on coroutines or any other kind of awaitable. This is kind of limiting. You can work around it by wrapping coroutine arguments in `fork` and wrapping awaitables in coroutine functions *and* `fork` 🤢. They also have crappy semantics. `Combine` returns nothing. `First` returns only the result of the first `await` to complete. There is no way of knowing *which* input fired. `First` also does not handle cancelling tasks that are passed in, which is very annoying when you are simply using tasks to work-around the fact that `First` doesn't take coroutines or awaitables.

I propose two new interfaces to replace `Combine` and `First` respectively: `gather` and `select`. `gather` will run any given awaitables concurrently and return a list of the results. `select` will run any given awaitables concurrently, and return once the first of the awaitables finishes. It will return the index (by position of the arguments) of the awaitable that fired, and the result.
Like `Combine` and `First`, exceptions will be propagated by default, and the awaitables that did not throw an exception will be cancelled safely.

```python
async def gather(*args: Awaitable[Any]) -> List[Any]:
    ...

async def select(*args: Awaitable[Any]) -> Tuple[int, Any]:
    ...
```

`gather` is extremely similar to `asyncio.gather`. There is no interface in `asyncio` for something like `gather`, but only for the first awaitable to finish. I decided on the word `select` because it is similar to the `select` statement in Go, albeit not just for queues.

Once these interface are implemented and tested, perhaps `Combine` and `First` should be deprecated. This is simply so that there is "one way to do it".

A follow-on may add an interface for *catching* exceptions as results rather than propagating them, much like `asyncio.gather` supports.

eric-wieser, 2021-08-11:
> There is no interface in `asyncio` for something like `gather`, but only for the first awaitable to finish.

Did you not find `asyncio.wait([a, b, c], return_when=FIRST_COMPLETED)`?

eric-wieser, 2021-08-11:
Also interesting is [this note in `asyncio`](https://docs.python.org/3/library/asyncio-task.html#asyncio.wait):

> Deprecated since version 3.8: If any awaitable in aws is a coroutine, it is automatically scheduled as a Task. Passing coroutines objects to wait() directly is deprecated as it leads to confusing behavior.

So arguably the behavior of forcing the user to `fork` first is already the right choice.

jwprice100, 2021-08-11:
Generally I'm all for this. Wanting to fork several coroutines, then wait for the first one to complete and kill the rest is a useful and common operation. I took what you gave me @ktbarrett and tweaked it to work for any coroutines not just queue.get(). This Select function is extremely useful and seems like a proper tool for cocotb. There may be debate about how it should be implemented but in general I think it would make sense.

I do have a concern though. What if two coroutines complete at the same simulation time. That is if one where to do:
`
idx, result = await Select(coro_a, coro_b) 
`
Is it not possible for coro_a to finish, return a result. Then as I understand it, the cocotb scheduler can now choose between running Select, or coro_b next. If coro_b runs first then Select, it would seem to me that it is possible coro_a and coro_b both ran and returned a result, but that one of them is lost.  Is this race condition possible, can it be avoided? In general in my use of the unofficial Select, I've avoided that possibility by design but is probably something to consider at some point.

ktbarrett, 2021-08-11:
> Is this race condition possible

Yes. Symmetric coroutines are not free from data races. This is expected.

> and if so can it be avoided?

I don't think it's possible.

ktbarrett, 2021-08-11:
@eric-wieser The "confusing behavior" is due to the fact that `wait` returns tasks rather than the result of the `await` like `gather` does. Not that `gather` does not have this problem. `select` is *not* like `wait`, but is more like `gather`.

They give this as the example of the confusing behavior:
```python
async def foo():
    return 42

coro = foo()
done, pending = await asyncio.wait({coro})

if coro in done:
    # This branch will never be run!
```

ktbarrett, 2021-08-11:
@jwprice100 Actually... I see your point. With the implementation of `select` I gave you it sets an `Event` and then continues. Another task could run before the `Event` is handled by `select` function. This is a bug and will not be present in the final version.

jwprice100, 2021-08-11:
@ktbarrett - What's cocotb's philosophy regarding deprecating/removing features. I agree on principal if gather/select are implemented than First/Combine are strictly inferior. You wouldn't want to have to maintain both or give developers multiple different options to accomplishing similar things, etc.  On the other hand those are used prolifically and would completely break many designs.

marlonjames, 2021-08-11:
cocotb has accumulated a lot of deprecations over the past few versions. We try not to break things on minor versions, so 2.0 should get rid of those, plus some other breaking changes that are sorely needed.

eric-wieser, 2021-08-12:
> This is not possible if you don't implement select with asyncio.wait like you have done. As soon as one task ends all others are unscheduled and cannot finish or fail.

This would no longer be true if we started supporting safe task cancellation (with cleanup).

> `wait` returning potentially many done tasks with `FIRST_COMPLETED` is not possible with cocotb's scheduler (and I think this only exists to handle `FIRST_EXCEPTION` anyways...).

Are you sure? If I call `wait` on two tasks that block on the same event, won't they both finish before `wait` is resumed? Or is that not how the scheduler sequences things

> Ergonomics is the problem we are trying to solve.

I was focusing on this problem from the OP:

> They also have crappy semantics. `Combine` returns nothing. `First` returns only the result of the first `await` to complete. There is no way of knowing _which_ input fired.

`wait` does not have these crappy semantics.

ktbarrett, 2021-08-12:
> won't they both finish before wait is resumed?

Depends... If you are `await`ing a Task and it finishes, execution jumps to the scheduler (due to the `yield self` in `RunningTask.__await__`); so you are rescheduling rather than bridging stacks. If you are awaiting a coroutine, no reschedule happens, the exception can be caught by the waiter task and the other tasks immediately killed before they are scheduled, even if they are waiting on the same trigger that fired. Everything comes down to how the scheduler works. Even if they are to go off "in the same time step" there is no guarantee in the current system 1 or more will finish. This really isn't a good place to be... We *can* guarantee one or the other. Seeing as awaiting coroutines bridges stacks and we can't change that, Tasks should do the same instead of rescheduling. In that case, we can guarantee there will be only 1 result.

> This would no longer be true if we started supporting safe task cancellation (with cleanup and Cancelled exceptions).

I guess this along with what I said means we don't really know until what a good interface is until we understand the semantics of the new scheduler (if that ever happens).

> I was focusing on this problem from the OP

There is the other mound of crap to deal with as well. I even added an emoji for effect!

>  You can work around it by wrapping coroutine arguments in `fork`, and wrapping awaitables in coroutine functions *and* `fork` 🤢

Personally, I run into this ergonomic problem more (others have as well, see #2494), and I have only run into `First`'s limitation now that I need to wait on multiple queues, which is why I'm finally proposing `select`. They are both problems, there is no reason to not fix both while we are at it.

ktbarrett, 2021-09-04:
There are issues with `asyncio.wait`. As soon as you segregate the futures into two different lists you obliterate any information matching the inputs to the outputs. Such a function should return a single list so that users can map outputs back to inputs.

Febbe, 2024-02-27:
What do you think about the idea of registering the cocotb scheduler as `asyncio.EventLoop` that would make cocotb's async functions compatible to asyncio, cocotb.task may also be replaced with asyncio.Task.

ktbarrett, 2024-02-28:
That has been proposed before. @marlonjames looked into it and IIRC came to the conclusion it wasn't feasible.

ktbarrett, 2025-12-19:
Closed by #5165.

closed 2025-12-19 by ktbarrett