cocotb/cocotb #2664: Add `FutureWarning` when a `fork`ed coroutine exception causes the test to end

ktbarrett, 2021-08-13:
I think we have decided that this behavior is not a good one, but instead we should capture the error in the Task's outcome and then throw a `RuntimeWarning` when the outcome is an exception, and it is never used before collection.

marlonjames, 2021-08-17:
xref #922

alexforencich, 2021-12-09:
So this means that an assert failure in a forked coroutine that is never awaited will not result in the test being terminated?  Is there a workaround for this to actually terminate the test in this case?

ktbarrett, 2021-12-09:
That is the other piece of the puzzle we haven't yet figured out. The current behavior will continue to work for 1.X, which will be here for a while longer.

alexforencich, 2021-12-09:
TBH, I think a better solution is more like adding "fork_with_exception_handling" or something along those lines.  If you're planning on awaiting and catching the exception there, then use that variant.  Otherwise, exception = test failure.  Silently throwing out exceptions and converting them to warnings is not a good idea and will break all kinds of existing testbenches.  If `fork` is deprecated anyway, then there should be no problem with leaving `fork` as-is, and setting up whatever the replacement is to improve on exception handling.

ktbarrett, 2024-08-13:
We have actually decided to go in another direction. Tasks started with `cocotb.start_soon` will behave as they did in cocotb prior to 2.0. There will be a new way of creating Tasks that if an exception is seen, will never fail the test, and will always store the exception until an `await` expression propagates it.

This allows us to retain existing behavior for backwards compatibility, while introducing new features that work as we intend and avoid the aforementioned issue.

With these new features we don't anticipate users `await`ing Tasks created with `cocotb.start_soon`, so run and forget Tasks for implementing things like Monitors, Drivers, Sequencers, etc. will continue to do "the right thing" by failing the test if those Tasks should fail.