cocotb/cocotb #2669: Problem with debugpy?

raysalemi, 2021-08-15:
When I try to get a host and port to listen for the debugger like this:

     listen_host, listen_port = debugpy.listen(("localhost", 5678))

I get this error:

     input in flex scanner failed

This was working recently.  Any notion as to what this error mean?

marlonjames, 2021-08-17:
Where is the error printed? Is it in a Python exception traceback?

raysalemi, 2021-08-17:
> Where is the error printed? Is it in a Python exception traceback?

No. It's just printed to the screen and the listen() never returns.

marlonjames, 2021-08-17:
I see you are using local host, so you could try `debugpy.listen(5678)` and see if that works.
The only thing I can find in a search is the GNU flex tool...

raysalemi, 2021-08-17:
Hi,

I got the same result with only the address. Here is a more complete set of the error messages:

```text
     0.00ns INFO     Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     Running tests with cocotb v1.5.2 from /Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1629234645
input in flex scanner failed
     0.00ns CRITICAL Failed to import module tinyalu_tb: timed out waiting for adapter to connect
     0.00ns INFO     MODULE variable was "tinyalu_tb"
     0.00ns INFO     Traceback: 
     0.00ns INFO     Traceback (most recent call last):
                       File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/server/api.py", line 218, in listen
                         sock, _ = endpoints_listener.accept()
                       File "/Users/raysalemi/opt/anaconda3/lib/python3.8/socket.py", line 292, in accept
                         fd, addr = self._accept()
                     socket.timeout: timed out
                     
                     During handling of the above exception, another exception occurred:
                     
                     Traceback (most recent call last):
                       File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/regression.py", line 175, in _discover_tests
                         module = _my_import(module_name)
                       File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/regression.py", line 66, in _my_import
                         mod = __import__(name)
                       File "/Users/raysalemi/repos/python4uvm_examples/TinyALU_04_uvm_test/tinyalu_tb.py", line 9, in <module>
                         listen_host, listen_port = debugpy.listen(5678)
                       File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/__init__.py", line 113, in listen
                         return api.listen(address)
                       File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/server/api.py", line 143, in debug
                         log.reraise_exception("{0}() failed:", func.__name__, level="info")
                       File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/server/api.py", line 141, in debug
                         return func(address, settrace_kwargs, **kwargs)
                       File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/server/api.py", line 232, in listen
                         raise RuntimeError("timed out waiting for adapter to connect")
                     RuntimeError: timed out waiting for adapter to connect
                     
Traceback (most recent call last):
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/server/api.py", line 218, in listen
    sock, _ = endpoints_listener.accept()
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/socket.py", line 292, in accept
    fd, addr = self._accept()
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/__init__.py", line 181, in _initialise_testbench
    return _initialise_testbench_(argv_)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/__init__.py", line 282, in _initialise_testbench_
    regression_manager = RegressionManager.from_discovery(top)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/regression.py", line 154, in from_discovery
    return cls(dut, tests, hooks)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/regression.py", line 126, in __init__
    for test in tests:
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/regression.py", line 175, in _discover_tests
    module = _my_import(module_name)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/regression.py", line 66, in _my_import
    mod = __import__(name)
  File "/Users/raysalemi/repos/python4uvm_examples/TinyALU_04_uvm_test/tinyalu_tb.py", line 9, in <module>
    listen_host, listen_port = debugpy.listen(5678)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/__init__.py", line 113, in listen
    return api.listen(address)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/server/api.py", line 143, in debug
    log.reraise_exception("{0}() failed:", func.__name__, level="info")
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/server/api.py", line 141, in debug
    return func(address, settrace_kwargs, **kwargs)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/debugpy/server/api.py", line 232, in listen
    raise RuntimeError("timed out waiting for adapter to connect")
RuntimeError: timed out waiting for adapter to connect
     0.00ns ERROR    cocotb initialization failed - exiting
     0.00ns ERROR    Failing test at simulator request before test run completion: Simulator shutdown prematurely
```

raysalemi, 2021-08-17:
Using the port alone made no difference. 

But I have some interesting additional data:

I created a tiny version of my test module:

```python
import cocotb

import debugpy
listen_host, listen_port = debugpy.listen(("localhost",5678))
print("LISTEN_HOST:", listen_host)
print("LISTEN_PORT", listen_port)
cocotb.logging.info("Waiting for Python debugger attach"
" on {}:{}".format(listen_host, listen_port))
# Suspend execution until debugger attaches
debugpy.wait_for_client()
# Break into debugger for user control
breakpoint()  # or debugpy.breakpoint() on 3.6 and below

@cocotb.test()
async def test(dut):
    print("TEST")
    assert True
```

If I run this module using Python, it works. It connects to the debugger and everything:

```bash
(base) raysalemi@RayProMac TinyALU_04_uvm_test % python test.py
LISTEN_HOST: 127.0.0.1
LISTEN_PORT 5678
```

I have this Makefile:

```make
CWD=$(shell pwd)
COCOTB_REDUCED_LOG_FMT = True
SIM ?= icarus
VERILOG_SOURCES =$(CWD)/hdl/verilog/tinyalu.sv
MODULE := test.py
TOPLEVEL := tinyalu
COCOTB_HDL_TIMEUNIT=1us
COCOTB_HDL_TIMEPRECISION=1us
include $(shell cocotb-config --makefiles)/Makefile.sim

cleanall: clean
	@rm -rf __pycache__
	@rm results.xml
```

If I use the Makefile I get the error:

```bash
(base) raysalemi@RayProMac TinyALU_04_uvm_test % make
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
MODULE=test.py TESTCASE= TOPLEVEL=tinyalu TOPLEVEL_LANG=verilog \
        /usr/local/bin/vvp -M /Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:77   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:240  in _embed_sim_init                 Python interpreter initialized and cocotb loaded!
     0.00ns INFO     Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     Running tests with cocotb v1.5.2 from /Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1629235653
input in flex scanner failed
```

I wonder what the Makefile does differently.

raysalemi, 2021-08-17:
I think this may be beyond the scope of this Issues list.

The debugger works properly if I run `make` from a terminal outside of VS Code.

Shall I close it?

raysalemi, 2021-08-17:
Turns out my Visual Studio Terminal was somehow corrupted.  I launched `code .` from the directory where I had launched the sim earlier, created a new terminal, and it worked.

closed 2021-08-17 by raysalemi