cocotb/cocotb #2671: Does cocotb recommend using .value?

raysalemi, 2021-08-16:
Is this the recommended way to get values?

```python
                self.cmd_mon_queue.put_nowait((int(self.dut.A.value),
                                               int(self.dut.B.value),
                                               int(self.dut.op.value)))
```

Or does one leave off `.value`

cmarqu, 2021-08-16:
In the docs, we recently switched to always use `.value` even when not strictly needed, because it reduces the frustration level of new users who might have to hunt for cases where they are missing it /when/ it's needed.

raysalemi, 2021-08-16:
So the recommendation is to always use it.  Thanks!

themperek, 2021-08-17:
I think it can be closed. If not please reopen.

closed 2021-08-17 by themperek