cocotb/cocotb #2672: Potential rerun/reproducibility problem when using TESTCASE with multi-test MODULE

walt3k, 2021-08-17:
Opening PR per @ktbarrett in discussion here: https://gitter.im/cocotb/Lobby

In short, cocotb sets random.seed at the start of running a MODULE in init.py but does not set the seed again at the start of each @cocotb.test().  If the MODULE had multiple tests in it and a user sets TESTCASE to rerun any test but the first one in MODULE, then it's possible the random behavior for the rerun test will be different than the original failure.

A few thoughts:   Resetting the seed back to RANDOM_SEED (or the randomly generated seed) before each test would not be an attractive solution because then the random behavior for each test could be identical for all MODULE tests.   Ideally, the "testcase seed" would be set to different value for each test and derived from/based on the RANDOM_SEED (or randomly generated seed).   On way to get such a seed could be to offset the main seed based on some algorithm based on the test currently running or in other words:   Testcase Seed = Main Seed + **f(testcase)**        If this were true then:

- Each test would get a unique seed hopefully producing different random behavior than the other tests
- When rerunning a failing test in a multi-test MODULE, the user would only need to set RANDOM_SEED to the same Main Seed which was used (likely randomly generated) for the original failure.

I am new to cocotb so I could be missing details but it appears MODULE test names are essentially unique in that cocotb seems to combine all @cocotb.test() names in the MODULE which are the same into one test, no matter where they appear in the testcase file.  If so, then one possibility for the **f(testcase)** offset would be to hash the testcase name to produce a unique number.  Granted, it's possible the same number might result from different test names but would be hopefully uncommon.

marlonjames, 2021-08-17:
Each `cocotb.test()` decorated function gets a unique `_id` that could be used, but setting `MODULE` to a list of modules would change things depending on the order.

walt3k, 2021-08-17:
> Each `cocotb.test()` decorated function gets a unique `_id` that could be used, but setting `MODULE` to a list of modules would change things depending on the order.

Interesting, I did not know that - thx!  I looked at decorators.py where I think the _id you are referring to is assigned and if so, it appears _id is an incrementally increasing number (+1) for each test.  In my experience, it's not uncommon for a verification methodology to extract and store failing tests for regression purposes.  If the _id was used to offset the seed and the failing test was not the first one in the MODULE file, then I think a user could not clean out the other passing tests in the MODULE file and only save the failing one(s) for future regressions.   It seems like it would be best if the seed offset used to create the testcase seed was not related to order/position in the original MODULE file.

ktbarrett, 2021-09-23:
I think the best solution would be to set a seed per test when the test is created. If you maintain the same MODULE in the same order (and you should, test modules can have side effects, changing the loading or the order can break things) then each time the RM loads the tests, they get the same seed based off the RANDOM_SEED. From that point running a particular test will run with that same seed.

ktbarrett, 2021-09-23:
An alternative might be to assign a test seed based on the RANDOM_SEED and the test name. Test names (with the module name) are unique. This would allow you to add or remove tests and have the same RANDOM_SEED work as long as the test isn't renamed or the test body changed.

walt3k, 2021-09-23:
Yes, that seems to work.  When we found this, we locally fixed the problem in regression.py by importing hashlib and adding the following code at line 302

       ```
 # Reseed the python random module with a unique testcase seed based on the test name
        # to ensure TESTCASE rerun reproducibility. 
        h = hashlib.new('shake_128')
        h.update(test.__qualname__.encode())
        offset=int((h.hexdigest(4)),16)
        testcase_seed=cocotb.RANDOM_SEED+offset
        random.seed(testcase_seed)
        self.log.debug("Seeding testcase=%s cocotb.RANDOM_SEED=%s Offset=%s Testcase_seed=%s" % (test.__qualname__,cocot
b.RANDOM_SEED,offset,testcase_seed))
```

ktbarrett, 2021-09-23:
One issue with using the name of the test is that tests generated by `TestFactory` are fragile. Add more values to a `TestFactory` option and the indexes shift. It's another argument for computing the name of `TestFactory`-generated tests based on unique representations of the argument value, like pytest does.

I don't think using the test name to set a test seed on creation breaks any existing reasonable interfaces, so I'll try to get this in soon.

ktbarrett, 2021-09-28:
It might also be nice to support specifying the `seed` in the decorator. So if you have a particular test that hits some failing corner case with some random seed, you can run it always with that seed rather than having to run everything with RANDOM_SEED and TESTCASE set to run that particular test. I guess that means each test should also print the seed it's using when it runs?

walt3k, 2021-09-28:
I assume the way cocotb users rerun their tests and set the seed properly is by extracting the "random_seed" property value out of the results.xml file.  The first thought which jumps to mind if there is a decorator which overrides the normal testcase seed setting of random_seed + hash(testname)  (which you will need to store so you can set the decorator of another test to that seed), then it certainly could be useful to record the original combo testcase seed as well as the random_seed we have today in the results.xml file or as you said, print it.  However, since we always use an automated batch system to run both random and regression tests, we usually find it helpful to not to have important rerun data like this in stdout/capture of stdout where we have to parse the value we want out and formatting could potentially change.  For this reason, I'd vote for storing seed values in results.xml or another well structured file like this.

A few other thoughts:  I supposed it's possible a test could randomly generate, say, a list of 20 fantastic numbers or data values which then the rest of the testcase uses to initialize, drive, or check values later in the test.  Using the testcase seed from another test could be useful if you wanted those same 20 values but then wanted to do some things slightly different.  What happens in our world more often than not is a particular **_sequence of values/timings_** results in an interesting model state and we want to extract that sequence to be used in other tests which use the same sequence but change everything else up.  For this reason, we've not found it useful to reuse the same seed for a whole test to seed another test but we definitely have classes/objects which generate random sequences who's behavior we want to extract and reuse.  The problem with doing such a thing when you have on psuedo-random number generator being used for all test case randomness is because any difference in the number of draws done to the generator will change the sequence generated by the interesting object/case we wanted to reuse.

For this reason our test bench tool has independent random number generators for these random sequence objects.  In the normal case the user just lets the tool randomly pick a seed to be used but it is recorded by object in our output files.  Then, if a user finds one or more sequences are interesting/useful, they can extract and set the seed for those sequence objects in their other tests.  This allows the new tests to reuse the interesting sequences but randomize everything else.  I don't know enough about cocotb and how random sequences/etc are generated to say whether such an approach makes sense here but I thought I'd mention it.

Finally, something which has gained popularity for our users over the past 5 years is what is called "reseeding."  Many of our simulation environments run for very long wall clock times and in many cases, the cost of teardown/restart is quite high.  We've instituted the idea/ability for users to request our test bench base code "reseed" the test bench and keep running. To be honest, I'm not sure what bugs/results have been discovered using this approach in the broader community; however, I mention it only because of your decorator idea.  I am probably skimming over many details but it seems plausible for cocotb testcases to capture/reuse the "interesting seeds" you thought of but rather than at a testcase level, perhaps at inner-testcase level, even if using one common random number generator.  Suppose cocotb had some sort of syntax for requesting a reseed at any point in the test or, for rerun, specifying the seed to be used for reseed at a particular point.  If this existed then perhaps a savvy user would structure their initial random testcases like so:
`@cocotb.test()
async def doRandomStuff(dut):
    """Testing random stuff"""
    <do initializations>
    request_reseed(tag="Location1")
    <generate random sequences and drive them>
     request_reseed(tag="Location2")
    <generate random sequences and drive them>
     etc...
`
Presumably the code between two reseed requests could be extracted and reused in subsequent tests by getting the "tag" seed value from results.xml or wherever and specifying as the "seed" argument to a call to request_reseed() in the new tests.

Perhaps it makes more sense if the user has generator/driver objects rather than serial code like this, I don't know, but I just wanted to share the basic idea.   The advantage of reseeding at points like this does the same thing this Issue is trying to address but rather than at a MODULE/TESTCASE level it's a TESTCASE/routine level or what have you.   If you were to still want the ability to use the whole testcase seed value for another testcase, you could still do it with this approach I think because you'd simply say something like

``@cocotb.test()
async def reuseOtherTestcaseSeedTest(dut):
    """Using the testcase_seed we got randomly in another good test (1234) to seed this entire test"""
    request_reseed(tag="StartOfTest", seed=1234)
    <do initializations>
    and so on....`
    
Something like that.   Again, I'm not sure how useful it is to reseed and entire test like this but I'm pretty sure being able to capture/reseed with more localized seeds is very powerful.

No pressure on any of this.  Just sharing some ideas/food for thought.  Thx!

PR #2721 (2021-11-10): Better support test reproducibility
Closes #2672. Pinging @walt3k.

Tests are now reproducible if:
* the test name and module does not change
* the test body does not change
* the test is "pure", meaning it's not affected by its environment (the state of the DUT, the state of the Python environment)
* the same `RANDOM_SEED` is used
closed 2021-11-10 by ktbarrett