cocotb/cocotb #2676: Replace distutils

cmarqu, 2021-08-18:
We should replace our usage of `distutils` soonish:  `DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives`

This is reported with Python 3.10, see https://github.com/cocotb/cocotb/pull/2675/checks?check_run_id=3364741779#step:25:199 for an example.

marlonjames, 2021-08-18:
Also mentioned by @ktbarrett here: https://github.com/cocotb/cocotb/pull/2634#discussion_r673386064

ktbarrett, 2021-08-18:
[Migration Advice](https://www.python.org/dev/peps/pep-0632/#id23) section of the PEP states we should use `packaging` for replacing `distutils.version`. However, I'd rather not have a dependency, and `packaging` does not even have a class like `LooseVersion`. We are either going to have to vendor in `distutils.version` or make something new.

The use of distutils with setuptools is not a problem, setuptools is maintaining a copy of distutils internally to continue support of  building C extensions. They hack `sys.modules` so we don't have to change imports either.

PR #2678 (2021-09-23): Replace deprecated distutils
distutils.version has been vendored in.

Closes #2676.
closed 2021-09-23 by ktbarrett