cocotb/cocotb #2679: Add `cancel()` to `RunningTask and deprecate kill()

marlonjames, 2021-08-19:


ktbarrett, 2021-08-20:
So the point is just the name change? I think if we are going to introduce "another way" to kill a task it should behave at least slightly differently.

marlonjames, 2021-08-20:
Yes the point is to change the name and refer to cancelling a task. Brings the language in line with `asyncio` and `trio`.

ktbarrett, 2021-08-20:
Are there problems with making `cancel` throw `CancelledError` right away? Is that a lot of work? If it is, perhaps we should emit a `FutureWarning` about it throwing `CancelledError` in 2.0?

marlonjames, 2021-09-03:
It would take some work, as `kill` just calls `_unschedule`, which doesn't run the `Task`'s coroutine again.

marlonjames, 2023-10-12:
`cancel()` was added in #2877.
Deprecating `kill()` is to-be-decided.

ktbarrett, 2023-10-12:
I think kill() should be deprecated and I think we should also work to support throwing CancelledErrors.

closed 2025-03-11 by ktbarrett