cocotb/cocotb #2682: Verilator, file too small (length=8) file 'Vtop__ALL.a' for architecture x86_64

kirknodeng, 2021-08-23:
cocotb version : 1.6.0-dev0
operating system : macos ( v10.15.7, catalina )
Simulator : Verilator 4.210-75 
Python : v3.8.5
log : 
mkdir -p sim_build
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module adder --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS "-Wl,-rpath,/usr/local/lib/python3.8/site-packages/cocotb-1.6.0.dev0-py3.8-macosx-10.15-x86_64.egg/cocotb/libs -L/usr/local/lib/python3.8/site-packages/cocotb-1.6.0.dev0-py3.8-macosx-10.15-x86_64.egg/cocotb/libs -lcocotbvpi_verilator" --trace --trace-structs --timescale 1ns/1ps /Users/kirknodeng/WORK/cocotb/examples/adder/tests/../hdl/adder.sv /usr/local/lib/python3.8/site-packages/cocotb-1.6.0.dev0-py3.8-macosx-10.15-x86_64.egg/cocotb/share/lib/verilator/verilator.cpp
make -C sim_build  -f Vtop.mk
g++  -I.  -MMD -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o verilator.o /usr/local/lib/python3.8/site-packages/cocotb-1.6.0.dev0-py3.8-macosx-10.15-x86_64.egg/cocotb/share/lib/verilator/verilator.cpp
g++  -I.  -MMD -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o verilated.o /Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o verilated_dpi.o /Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/verilated_dpi.cpp
g++  -I.  -MMD -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o verilated_vpi.o /Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/verilated_vpi.cpp
g++  -I.  -MMD -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o verilated_vcd_c.o /Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/verilated_vcd_c.cpp
/opt/local/bin/perl /Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root__DepSet_h6944321b__0.cpp Vtop__Dpi.cpp Vtop__Trace__0.cpp Vtop___024root__Slow.cpp Vtop___024root__DepSet_h4b98a05d__0__Slow.cpp Vtop___024root__DepSet_h6944321b__0__Slow.cpp Vtop__Syms.cpp Vtop__Trace__0__Slow.cpp > Vtop__ALL.cpp
g++  -I.  -MMD -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include -I/Users/kirknodeng/local/verilator/v4.210-75-gf9806595/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o Vtop__ALL.o Vtop__ALL.cpp
echo "" > Vtop__ALL.verilator_deplist.tmp
Archive ar -rcs Vtop__ALL.a Vtop__ALL.o
ar: @Vtop__ALL.a.verilator_deplist.tmp: No such file or directory
g++    verilator.o verilated.o verilated_dpi.o verilated_vpi.o verilated_vcd_c.o Vtop__ALL.a   -Wl,-rpath,/usr/local/lib/python3.8/site-packages/cocotb-1.6.0.dev0-py3.8-macosx-10.15-x86_64.egg/cocotb/libs -L/usr/local/lib/python3.8/site-packages/cocotb-1.6.0.dev0-py3.8-macosx-10.15-x86_64.egg/cocotb/libs -lcocotbvpi_verilator    -o Vtop
ld: file too small (length=8) file 'Vtop__ALL.a' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [Vtop] Error 1
rm Vtop__ALL.verilator_deplist.tmp
make[1]: *** [sim_build/Vtop] Error 2
make: *** [sim] Error 2

When I tested both "adder" and "simple_dff" with latest verilator, I got this message above. 
So, I tried to test manually. I go to the "sim_build" directory. then I make "Vtop_All.verilator_deplist.tmp" manually. and I re-archive and link. I compiled and made a executable "Vtop" in "sim_build" directory. But, This executable is not work. 

Anyone know about this kind of situation? 

Thanks,

imphil, 2021-08-23:
Two comments:
- Please use Verilator 4.106, it's the only version that is known to work with cocotb at the moment.
- The error you're seeing is something I haven't seen before and most likely a problem in Verilator's Makefile on your machine. (I know there's limited testing on macOS).

So can you do two things?
- File a verilator bug at https://github.com/verilator/verilator for the makefile issue? Please run `make --warn-undefined-variables SHELL='sh -x'` instead of the make command you're running currently to give more logging output.
- For use with cocotb, use Verilator 4.106.

kirknodeng, 2021-08-25:
@imphil 
Thank you for your comments. 

Yes, I tried to test with Verilator 4.106. And, I changed "g++" version to 8.x. (I don't it is effected to test.. ) 
But, It's working right now..  
I'll close this issue. 

Thanks

closed 2021-08-25 by kirknodeng