cocotb/cocotb #2686: modelsim

Nagatejapallapu, 2021-08-30:
Hii,
while running the make helloworld.vsim i got an issue
[image]

Nagatejapallapu, 2021-08-30:
[image]

imphil, 2021-08-30:
The error message you posted at https://github.com/cocotb/cocotb/issues/758#issuecomment-908160844 indicates that you're missing 32 bit glibc packages:

> Warning: (vsim-7032) The 32-bit glibc RPM does not appear to be installed on this machine.  Calls to gcc may fail.

Have you fixed this problem in the meantime? An approach like https://stackoverflow.com/questions/49849001/are-32bit-libraries-not-supported-on-ubuntu-16-04 should help.

Nagatejapallapu, 2021-08-30:
I'm tried it but its not run,

collect2: error: ld returned 1 exit status
# 
# No such file or directory. (errno = ENOENT)
# 
# 
# FATAL ERROR while loading design
# Error loading design
# End time: 15:09:23 on Aug 30,2021, Elapsed time: 0:00:00
# Errors: 1, Warnings: 2

Nagatejapallapu, 2021-08-30:
[image]

Nagatejapallapu, 2021-08-30:
[image]