cocotb/cocotb #2688: Automatically deduction of TOPLEVEL_LANG don't work with Xcelium

rodrigomelo9, 2021-09-01:
Hi. It is my first time using cocotb with Xcelium (always GHDL or iverilog). Has far I understand it must work, but fail.

* the cocotb version used: 1.5.2
* the operating system and version: Centos7, 64 bits
* the simulator and version: Xcelium (19.09), 64 bits
* the Python version, and where it's coming from: 3.7.0 (pyenv)

Log:
```
/.../.pyenv/versions/3.7.0/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.xcelium:103: *** "A valid value (verilog or vhdl) was not provided for TOPLEVEL_LANG=".  Stop.
```

Makefile 
```
#TOPLEVEL_LANG=verilog

VERILOG_SOURCES += ../vlog/counter.v
TOPLEVEL = counter
MODULE = counter

include $(shell cocotb-config --makefiles)/Makefile.sim

clean::
	@rm -fr __pycache__ *.vcd *.xml
```

I need to uncomment TOPLEVEL_LANG. to avoid this.

ktbarrett, 2021-09-01:
What is the error? I don't have Xcelium available to me to test this. Is this breaking for other simulators?

rodrigomelo9, 2021-09-01:
First, let me say that is a minor issue. Can be easily fixed by assigning a value to TOPLEVEL_LANG.

I wrote the error in the issue description:
> Makefile.xcelium:103: *** "A valid value (verilog or vhdl) was not provided for TOPLEVEL_LANG=".  Stop.

From the release notes of cocotb 1.5.0, I read:
> Makefiles now automatically deduce TOPLEVEL_LANG based on the value of VERILOG_SOURCES and VHDL_SOURCES. Makefiles also detect incorrect usage of TOPLEVEL_LANG for simulators that only support one language. (#1982)

Maybe I miss understood?

It works with GHDL and Icarus, but these are not fair examples :P I haven't access to other simulators.

ktbarrett, 2021-09-01:
If you have both Verilog and VHDL sources, you have to tie break. Here is the logic. It only deduces VHDL or Verilog is the opposite `SOURCES` variable is empty or not set. The Makefiles are not smart enough to determine the language of the top. https://github.com/cocotb/cocotb/blob/f55502d013b7c7035180750755cede36844d8d7c/cocotb/share/makefiles/Makefile.inc#L111-L123

rodrigomelo9, 2021-09-01:
But I have only Verilog files... one Verilog file to be more precise. I wrote also mi Makefile in the issue description:

```
#TOPLEVEL_LANG=verilog

VERILOG_SOURCES += ../vlog/counter.v
TOPLEVEL = counter
MODULE = counter

include $(shell cocotb-config --makefiles)/Makefile.sim

clean::
	@rm -fr __pycache__ *.vcd *.xml
```

I didn't set VHDL_SOURCES.

ktbarrett, 2021-09-02:
So I added a print at line 112 above and it started working. Seems like it's a gmake bug...