cocotb/cocotb #2699: Need help with fixing the cocotb code as outputs are always 'x'

varain007, 2021-09-08:
Dear all,

I have a system verilog code for an adc. The simulation works fine on edaplaygrounds and am able to see outputs. To verify the design, I created a sample cocotb code and am new to it. However, the outputs from running the python code always gives x in the outputs. 

Any help would be great. I have attached the code for your reference.

Thanks,
[adc.zip](https://github.com/cocotb/cocotb/files/7127700/adc.zip)

ktbarrett, 2021-09-08:
Please provide your simulator, simulator version, and operating system.

varain007, 2021-09-08:
I'm using icarus 2.3.1 and ubuntu 21.04

ktbarrett, 2021-09-08:
That version number doesn't seem correct. It should probably be 10.X or 11.0?

varain007, 2021-09-08:
Sorry about that. It's 11.0

varain007, 2021-09-08:
I can provide the source code if required

ktbarrett, 2021-09-08:
Did you not already provide the source code in the original post? I see a zip file with sources and the python test (but no Makefile). I didn't see anything obvious, so I won't be able to help debug until I get off work tonight.

varain007, 2021-09-08:
Yes I just remembered I already gave the source files.
Here is the make file too 

SIM ?= icarus
TOPLEVEL_LANG = verilog
VERILOG_SOURCES = $(shell pwd)/adc_ctrl.sv
TOPLEVEL = adc_ctrl
MODULE = test_adc
include /home/varun/cocotb/cocotb/share/makefiles/Makefile.sim

varain007, 2021-09-09:
@ktbarrett were you able to find something?

Thanks

sjalloq, 2021-09-09:
If you reset your design it might help.

varain007, 2021-09-09:
Well i did reset it. Now all outputs are zero.

sjalloq, 2021-09-09:
So that's fixed this issue.  ;-)  If your outputs are not supposed to be zero, then debug further...

varain007, 2021-09-09:
Well, the actual issue is that there is no output with cocotb. But it runs properly with a normal verilog tb

varain007, 2021-09-09:
@sjalloq Could you help if you have experience writing the cocotb

sjalloq, 2021-09-09:
...or just run longer...

```python
import cocotb, logging
from cocotb_coverage.coverage import *
from cocotb.triggers import Timer
from cocotb.triggers import FallingEdge, RisingEdge, ClockCycles
from cocotb.clock import Clock
import random
import matplotlib.pyplot as plt
import numpy as py
import xlsxwriter
import itertools

class TB:
  def __init__(self, dut):
    self.dut   = dut
    self.clk   = dut.clk_i
    self.rst_n = dut.rst_n_i
    self.log   = logging.getLogger(type(self).__qualname__)
    self.log.setLevel(logging.DEBUG)

    self.dut.adc_ch_disable_i.setimmediatevalue(0)
    self.dut.adc_conv_time_i.setimmediatevalue(0)
    self.dut.di_adc_comp_out_i.setimmediatevalue(0)
    self.dut.adc_ctrl_en_i.setimmediatevalue(0)
    self.dut.adc_sample_time_i.setimmediatevalue(0)
    self.dut.adc_vbat_sample_set_i.setimmediatevalue(0)
    self.dut.adc_cs_sample_set_i.setimmediatevalue(0)
    self.dut.adc_therm_sample_set_i.setimmediatevalue(0)
    self.dut.adc_vboost_sample_set_i.setimmediatevalue(0)
    self.dut.adc_tst_sample_set_i.setimmediatevalue(0)

    cocotb.fork( Clock(self.clk, 100, units='ns').start() )

  async def reset(self):
    self.rst_n <= 0
    await Timer(200, 'ns')
    await RisingEdge(self.clk)
    self.rst_n <= 1

@cocotb.test()
async def adc_random_test(dut):

  tb = TB(dut)
  await tb.reset()

  col = 0
  row = 0

  dut.adc_ch_disable_i        <= 10
  dut.adc_conv_time_i         <= 2
  dut.di_adc_comp_out_i       <= 1
  dut.adc_ctrl_en_i           <= 1
  dut.adc_sample_time_i       <= 1
  dut.adc_vbat_sample_set_i   <= 1
  dut.adc_cs_sample_set_i     <= 0
  dut.adc_therm_sample_set_i  <= 2
  dut.adc_vboost_sample_set_i <= 3
  dut.adc_tst_sample_set_i    <= 4

  for _ in range(200):
    await RisingEdge(dut.clk_i)
    print("dutclk, disable, conv, sample, vbat, array, vin, data:",dut.clk_i, dut.adc_ch_disable_i, dut.adc_conv_time_i, dut.adc_sample_time_i, dut.adc_vbat_sample_set_i, dut.do_adc_array_o, dut.do_adc_vin_sel_o, dut.adc_data_vbat_o)

```

varain007, 2021-09-09:
Okay thanks for your code @sjalloq  :) I tried it however I get the following error for all the 200 times it iterates.
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.9/logging/__init__.py", line 1083, in emit
    msg = self.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 927, in format
    return fmt.format(record)
  File "/home/varun/.local/lib/python3.9/site-packages/cocotb/log.py", line 248, in format
    msg = record.getMessage()
  File "/usr/lib/python3.9/logging/__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/home/varun/.local/lib/python3.9/site-packages/cocotb/scheduler.py", line 358, in _react
    self._event_loop(trigger)
  File "/home/varun/.local/lib/python3.9/site-packages/cocotb/scheduler.py", line 452, in _event_loop
    self._schedule(coro, trigger=trigger)
  File "/home/varun/.local/lib/python3.9/site-packages/cocotb/scheduler.py", line 857, in _schedule
    result = coroutine._advance(send_outcome)
  File "/home/varun/.local/lib/python3.9/site-packages/cocotb/decorators.py", line 317, in _advance
    return super(RunningTest, self)._advance(outcome)
  File "/home/varun/.local/lib/python3.9/site-packages/cocotb/decorators.py", line 201, in _advance
    return outcome.send(self._coro)
  File "/home/varun/.local/lib/python3.9/site-packages/cocotb/outcomes.py", line 36, in send
    return gen.send(self.value)
  File "/home/varun/adcctrl/test_adc.py", line 61, in adc_random_test
    tb.log.info("dutclk, disable, conv, sample, vbat, array, vin, data:",dut.clk_i, dut.adc_ch_disable_i, dut.adc_conv_time_i, dut.adc_sample_time_i, dut.adc_vbat_sample_set_i, dut.do_adc_array_o, dut.do_adc_vin_sel_o, dut.adc_data_vbat_o)
Message: 'dutclk, disable, conv, sample, vbat, array, vin, data:'
Arguments: (ModifiableObject(adc_ctrl.clk_i), ModifiableObject(adc_ctrl.adc_ch_disable_i), ModifiableObject(adc_ctrl.adc_conv_time_i), ModifiableObject(adc_ctrl.adc_sample_time_i), ModifiableObject(adc_ctrl.adc_vbat_sample_set_i), ModifiableObject(adc_ctrl.do_adc_array_o), ModifiableObject(adc_ctrl.do_adc_vin_sel_o), ModifiableObject(adc_ctrl.adc_data_vbat_o))

sjalloq, 2021-09-09:
Sorry, I updated the log line after you copied it

varain007, 2021-09-09:
@sjalloq Thank you :) It works now. Am able to see some outputs. However, i can see it can no longer run more than 20,600 ns for whatever reason. Do you know if it can be made to run any longer? If the iteration is more than 200 it just hangs. Maybe it's a simulator thingy?

sjalloq, 2021-09-09:
It works fine in Xcelium so likely an iverilog issue.  You could try putting it through sv2v and see if iverilog still hangs.

varain007, 2021-09-09:
Okay. That's good news that it works with Xcelium. I shall try navigating the code to verilog and see

varain007, 2021-09-09:
@sjalloq Could you share your simulation outputs with xcelium?

sjalloq, 2021-09-09:
Sorry, you're on your own from here.....  You could open a ticket against iverilog to see if they can help debug it.

closed 2021-10-17 by github-actions[bot]