cocotb/cocotb #2702: Release cocotb 1.6

imphil, 2021-09-08:
Milestone: https://github.com/cocotb/cocotb/milestone/18

Release process checklist:
- [ ] Close all issues blocking the milestone
- [ ] prepare RC release
- [ ] prepare release announcement email
- [ ] tag release
- [ ] test release build on PyPi
- [ ] send out announcement to mailing list

marlonjames, 2021-09-10:
There are a few `versionchanged` and `versionadded` references to 2.0 in the docs that should be converted 1.6

imphil, 2021-09-29:
rc1 release is done and uploaded to pypi.

imphil, 2021-10-17:
rc2 is now building. We'll give this a last round of testing and I expect to release 1.6.0 final which is identical to rc2.

imphil, 2021-10-20:
Release is published. We're done here!