cocotb/cocotb #2706: To Join or not to Join

raysalemi, 2021-09-11:
I don't see a difference between awaiting the coroutine returned by `fork` and awaiting `Join` using the coroutine returned by `fork`.

The two approaches return the same data and seem to work the same way:

```python
@cocotb.test()
async def awaiting_a_fork(dut):
    """Learning test to see how awaiting a fork works"""
    c1 = cocotb.fork(counter("Counter 1", 1, 3))
    cr = await c1
    logger.info(f"{cr} done counting")
    c2 = cocotb.fork(counter("Counter 2", 1, 3))
    cr = await cocotb.triggers.Join(c2)
    logger.info(f"{cr} done counting")
```

Results are identical:

```text
     4.00ns INFO     Starting test: "awaiting_a_fork"
                     Description: Learning test to see how awaiting a fork works
     5.00ns INFO     Counter 1 counts 1
     5.00ns INFO     Counter 1 done counting
     6.00ns INFO     Counter 2 counts 1
     6.00ns INFO     Counter 2 done counting
     6.00ns INFO     Test Passed: awaiting_a_fork
````

marlonjames, 2021-09-12:
`await`ing a task is converted to a `Join` by the scheduler automatically, so they are equivalent.

closed 2021-09-12 by raysalemi

ktbarrett, 2021-09-12:
That being said you should prefer `await`, it is less likely to be deprecated in the future than `join`.

We don't like synonyms, "There should be one– and preferably only one –obvious way to do it." At some point that will be resolved.

raysalemi, 2021-09-12:
Yes. I came to the same conclusion.