cocotb/cocotb #2719: Add stdout of the test logger to xUnit output

ktbarrett, 2021-09-23:
This was removed accidentally a while ago and now there is no way to determine what argument values TestFactory were passed, or the stacktrace that caused the failure. xUnit purposefully supports capturing stdout and I think the best way of accomplishing that is by capturing the log messages emitted by the `cocotb.*` namespace logger, or maybe just the logger specifically for the test object.

tymonx, 2025-12-27:
This will require to add a capture mechanism for logger similar to what pytest is doing. Registering a log handler that will tee content of logged messages to separate string buffer `StringIO` and dump them to `system-out` and `system-err` XML elements of test case.

Currently out of scope of this PR https://github.com/cocotb/cocotb/pull/5219 but I will try to keep this in mind and prepare a separate PR 🤔

ktbarrett, 2025-12-27:
Let's close this and track #2548.

closed 2025-12-27 by ktbarrett