cocotb/cocotb #2724: Simulation restart on Questa fails to restart CocoTB(cocotb.gpi already registered)

aaelmaghl, 2021-09-28:
This issue appears to be independent of the Questa, Python or cocotb versions used.
I have a simple RTL simulation that uses cocotb running on Questa/Modelsim. On the simulation's conclusion I call vsim's restart function and am met with the following log message

> 200.00ns WARNING cocotb.gpi GpiCommon.cpp:104 in gpi_register_impl VPI already registered, Check GPI_EXTRA

Upon rerunning the simulation(run -all), the system-verilog side of the simulation runs fine but the cocotb/python/pli side of things appears to break. None of the statements in the python testbench appear to execute at all, no log messages or HDL forces/reads and the system verilog RTL progresses as if the simulation never even uses CocoTB.

I've looked around and it appears someone else has had the issue(apologies if I should just have posted there instead, I'm new to posting on github and I'm not sure on what the guidelines for this sort of thing are). 
https://github.com/cocotb/cocotb/issues/284
However it hasn't been updated in two years and I'm not sure what the progress or status of that issue is.

Thanks.

sjalloq, 2021-09-28:
I had this issue and reported it to Mentor in my last job (Cadence house now).  I have no idea if they ever fixed this but I doubt it.  Mentor's tools have fallen very far in terms of quality in the last 10 years.

The work around to this is to just run the simulations in batch mode and reload the wave database if you're trying to debug waves.  You can't use the restart functionality with Questa.

raysalemi, 2021-09-28:
FOSSi is a Siemens technical partner, so if this is a bug in Questa we have an avenue to open a service request.  However it's not clear to me yet that this is a Questa bug.

ktbarrett, 2021-09-28:
This was never supported by cocotb, it's a missing feature. It's unknown if Questa is misbehaving as no one has explored adding this feature. This issue is a duplicate as you stated.

closed 2021-09-28 by ktbarrett

raysalemi, 2021-09-28:
Sounds like Xcelium got lucky.

raysalemi, 2021-09-28:
> I'm fairly certain that is a happy accident. If the restart runs the shutdown, then the start hooks, it works. That behavior is probably wrong (to your benefit), it should probably work more like Questa and only run the restart hook. The VPI standard is extremely vague.

<shakes fist> Xcelium!!!!!

ktbarrett, 2021-09-28:
I'm not trying to cast too many stones without better understanding what the problem is, but I'm saying it's just as likely that Xcelium is misbehaving as Questa and no one knows because no one has done any research on the topic.

raysalemi, 2021-09-28:
Given that this is not a feature of **cocotb** the "happy accident" theory makes the most sense.

sjalloq, 2021-09-28:
Well, whatever the reasons, when I reported it to Mentor, they acknowledged an issue and opened an internal ticket.  No way of tracking that now though so not much use.

raysalemi, 2021-09-28:
@sjalloq , could you give me the SR number?  I can look into it.

marlonjames, 2021-09-28:
This is a complicated issue, and really depends on the interactions between the dynamic libraries (cocotb as well as Python and any loaded modules that use c extensions) and the simulator.
I explored this for doing restart in the Active-HDL GUI and was never able to get things working correctly.

Although it has been improved, the GPI layer still makes assumptions about cocotb running a single time through and shutting down, so the simulator exiting does the full cleanup. Re-running from the initial bootstrap function may leak VPI/VHPI handles and callbacks, and the embedded Python interpreter may not work correctly the second time through.

I saw issues with `Py_Finalize` followed by `Py_Initialize` in the same thread, which is a known issue, where not all of the loaded Python module extensions (DLLs) were unloaded. I also saw issues where Active-HDL would change the stdin/stdout/stderr file descriptors between runs. Another issue is depending on how the simulator is designed, it may not be possible to change the environment variables between test runs.

raysalemi, 2021-09-28:
I think if **cocotb** was not designed to restart then it can't be expected to do the restart, even if one simulator accidentally allows it in some cases.

mvsoliveira, 2021-10-09:
Hi, there is something that I don't quite understand. On the first page of cocotb docs, I read the following:
"Python is interpreted - tests can be edited and re-run without having to recompile the design or exit the simulator GUI"
But in this issue, I read that restarting the simulation is not supported. Am I missing something?

mvsoliveira, 2021-10-09:
I am not sure if I understand completely the difference from run from scratch or setting the simulation back to 0 and run it again. I would like to be able to change the python code and run the simulation again without exiting the Questa GUI. Is it possible?

raysalemi, 2021-10-09:
The question is whether you type the vsim command again and reelaborate the
design. I believe you can do that in the GUI,  but not by presssing the
restart button.

raysalemi, 2021-10-09:
> Technical issues with reseting the time back to 0 and how that's supported in the VPI/VHPI/FLI. Letting time go forward and reseting the dut and running new tests is a simpler operation to accomplish.

How does resetting the DUT incorporate modifications to the Python in the testbench?

raysalemi, 2021-10-09:
> But that statement you quoted is false, it isn't supported _currently_ on many simulators. I'm not saying it's out of scope, just that someone needs to do the work to get it to work.

If I read this correctly, you're saying that people did the work for other simulators but not for Questa?  So the capability exists for Questa if someone wants to implement it?  True?

mvsoliveira, 2021-10-10:
So just for the record, it works by retyping the vsim command automatically generated in the rumsim.do file. 
If one does not care about compiling the HDL files again, one can just run runsim.do file again after the simulation is finished. The GUI wont be closed and modifications in the python testbench or HDL files will be taken into account.

gbowers4, 2022-03-15:
> Technical issues with reseting the time back to 0 and how that's supported in the VPI/VHPI/FLI. Letting time go forward and reseting the dut and running new tests is a simpler operation to accomplish.

Regarding the above - is there a reccomended approach or open issues wrt to resetting the dut in between tests - or is some work required to implement this

riteshmentor, 2022-03-15:
@gbowers4 The last reply from @mvsoliveira is the suggested solution from Questa point of view. This is actually called "reinvoke" of the simulator. "Reinvoke" is supposed to launch a new session for the user, where all the initialization, library loading happens from scratch. At Questa, we had debugged issues related to "reset", which is essentially moving the simulation time to ZERO. It appears that later versions of CocoTb had implemented some Python initialization and finalize routines which are interfering with doing a "reset".

marlonjames, 2022-03-15:
Does Questa support the reset callbacks?

- VPI
  - `cbStartOfReset` and `cbEndOfReset`
  - Reset can be achieved with `$reset` system task or `vpi_control(vpiReset, ...)` and other means not defined in the standard
- VHPI
  - `vhpiCbStartOfReset` and `vhpiCbEndOfReset`
  - Reset can be achieved with `vhpi_control(vhpiReset, ...)` and other means not defined in the standard

If these callbacks were supported, it would allow cocotb to distinguish between a reset and a shutdown, and the issues with finalizing the interpreter could be avoided.